hamish2014/FreeCAD_assembly2 #1: Great Job! - a couple of issues/requests.

temecom, 2015-01-02:
I was just saying: 'FreeCAD is missing a good assembly module' and then I found this. Very nice first release. A couple of things are missing or would be nice: 

1. Constraints don't seem to be active during transform mode. 
2. Plane to Plane constraints don't seem to work on circular planes. eg: a ring seated at the bottom of a circular pocket. 
3. Refresh part changes after loading... Is there a way to do this?

Thanks

Chris

temecom, 2015-01-03:
Ignore # 2 above. They do work I was using the axial rather than circular constraint. 

Please add: 4. There needs to be a way to reverse an assembly constraint. The engine picks one mating mode and I don't see a way to flip to the opposite mode without introducing an alignment constraint first.

temecom, 2015-01-03:
Ok so I figured out all of the above except #1. 

3. The Update button refreshes the parts. (I originally thought that this updated the part... duh!)
4. The opposed/aligned setting is in the constraint direction property

Very nice indeed !

hamish2014, 2015-01-07:
Thanks temecom.
I am very glad you are finding the assembly2 workbench (wb) useful.
If you find any other bugs, please do as you have done, and register a issue on the github page.

closed 2015-01-07 by hamish2014