hamish2014/FreeCAD_assembly2 #2: Part updating sometimes breaks

hamish2014, 2015-01-07:
Parting updating often breaking constraints, due to the faces/edges being renumber when that part is updated.
I am aware of this and it is on the top of my assembly2 to-do list.

triplus, 2015-01-07:
Hi.

I am not sure but i am guessing until topological naming is implemented in FreeCAD:

http://www.freecadweb.org/wiki/index.php?title=Naming_project

It will be hard to achieve robust solution. Maybe i misunderstood this bug report or maybe you have some workaround how to improve situation with your assembly2 WB available. I just mentioned topological naming because there are other bugs and "bugs" in FreeCAD and i guess not much can be done about them until robust topological naming is implemented.

hamish2014, 2015-01-09:
Thanks triplus, you did understand the bug correctly.
When i posted the issue, i was thinking of a crude topological solution to repair damaged constraints.
But this crude topological solution is going to rather difficult to implement.
My thoughts at the moment are therefore to enhance functionallity to repair/redine constraint  so that users can correct these errors themselves in an easier manner.

triplus, 2015-01-09:
Enhancing functionality to redo constraints seems like a good plan.

hamish2014, 2015-01-21:
Redefine constraint functionality has now been added.
Now if an part update does go wrong, users can redefine the problem constraints using the context/pop-up Menu in FreeCAD tree view, under constraint utilities.

triplus, 2015-01-23:
Hi!

I tested new redefine constraint functionality. I imported two simple objects to assembly and after added plane constraint between them. I intentionally changed geometry of one of the imported objects and after assembly update operation plane constraint flipped to different face (as expected). Using context menu to redefine plane constraint worked as expected! I would say unless you will work on topological naming you could close this bug report because this is not something you have control over as this is something to implement in FreeCAD!

Side note:

https://github.com/hamish2014/FreeCAD_assembly2/issues/3

Placing imported object using mouse operation works great!

Cheers!

hamish2014, 2015-01-24:
Agreed, closing issue.

hamish2014, 2015-04-29:
Update part command updated 1 week back.

Update, entailed matching subelements from the parts new shape against the parts old shape, as to prevent things breaking.

matching code still likely to have some bugs ...