hamish2014/FreeCAD_assembly2 #4: Plane constraint and faces not on XY/XZ/YZ plane

triplus, 2015-01-26:
Hi.

I was wondering if scenarios when inserting plane constraint should work if selected face of one of the objects is not parallel with XY/XZ/YZ plane. It works fine when selected face is parallel but i didn't manage to get working scenario when the face is rotated at a angle to XY/XZ/YZ plane.

Cheers.

hamish2014, 2015-01-27:
Plane constraints should work for any orientation, not just the for XY/YZ/ZX cases.
Please send me a FreeCAD file with such an assembly constructed, and will try have a look at what is causing the solver to crash.

triplus, 2015-01-27:
Hi.

You are correct it does work sometimes but i still get this quite often:

> The assembly2 solver failed to satisfy the constraint "planeConstraint01".

I didn't attach the file because i get random results.  Try this instead and create 2 objects as seen on the image and import them and try to add plane constraint. Always select the angled face on triangle object but on cube select random face:

[image]

After few tries you should be able to see mentioned solver message and investigate what exactly is happening.

Cheers!

hamish2014, 2015-01-27:
Okay, I added a series of test cases the latest version of the assembly2 wb.
Using these test cases I managed to find and patch some bugs.
The random behaviour you referred to, resulted from float numerical errors which caused arccos to bomb out: i.e. arccos( 1 + 10^-32 )

Please check if everything works on your side with the latest version.

triplus, 2015-01-27:
I can confirm it works fine now.

Thanks.

closed 2015-01-27 by hamish2014