hamish2014/FreeCAD_assembly2 #5: Constraint surface whit surface no possible

corvusd, 2015-01-27:
First of all thank you for you great work.

The actual state of the module is great.

If possible implement, as improved critical, constraint between faces of the body´s?

I have used these objects to my tests: https://drive.google.com/file/d/0B0mSen1pNtD4aEFXVENhb2dCd1U/view?usp=sharing

The test consist in position the object like this example: http://www.iearobotics.com/wiki/images/thumb/e/e7/T39-ej1-soporte-aquaris-5-7-3.png/300px-T39-ej1-soporte-aquaris-5-7-3.png

If you want or need that I help you to improve your project. Only tell me something via email: david.gamiz@gmail.com

Greetings,

David Gámiz Jiménez

hamish2014, 2015-01-27:
Hi David, 

Thank you for registering the issue.
I have downloaded the files, and will see what i can do.

corvusd, 2015-01-27:
Thanks again.

For anything you need, here I am.

I will continue testing your module. :)

hamish2014, 2015-01-28:
Hi David,

Okay, so the difficulties you encountered when trying to assemble the phone+stand originated from lack of flat surfaces.
Almost all the surfaces on the phone are curved, very slightly.
To overcome this, i ended up adding functionality to the assembly 2 workbench.
Functionality added:
 - plane-vertex constraints
 - edge-edge angle constraints

[image]

corvusd, 2015-01-28:
PLAS PLAS! applauses! :D :D

Took more than eight years in the free software community.

And I never cease to amaze me. Rejoice in the great people there!

Thank You. I continue by testing your module. And bringing everything I can.

A strong greeting! : D

hamish2014, 2015-01-29:
Pleasure, keep well.

closed 2015-01-29 by hamish2014