hamish2014/FreeCAD_assembly2 #6: Followed Ubuntu installation instructions. Workbench is selectable, but throws error when selected.

TimothyStiles, 2015-02-07:
Hello!

Thanks for making this module. I am really excited to try it and followed the ubuntu installation instructions, but I have run into a problem. When I load freecad it gives me the option to select "assembly 2." When I select the option it throws me the error in the screenshot attached and then does not provide any tools in the workbench. Any thoughts as to what could be going wrong? Thanks!

[image]

normandc, 2015-02-07:
What version of Ubuntu and of FreeCAD are you using? The Assembly2 module requires FreeCAD 0.15 which is currently in bug fixing phase for a planned release at the end of February. But the development version is already quite stable, you can update your FreeCAD by adding the Daily Builds PPA to your software sources. See http://www.freecadweb.org/wiki/index.php?title=Download#Ubuntu_PPA_packages

With FreeCAD 0.15.4537 on Ubuntu 14.04 LTS 64-Bit, the Assembly2 module works as expected.

normandc, 2015-02-07:
Also, you may get more useful output by enabling the Report view (under the View --> Views menu), going to Edit --> Preferences / General / Output and checking both "Redirect internal python messages..." and "Redirect internal python errors to the report view".

TimothyStiles, 2015-02-07:
I have the same FreeCAD version, but I have Ubuntu 14.10 64 bit without the long term support. I recemtly installed Ubuntu specifically so I could help develop FreeCAD. Should I reinstall with long-term support if I do not want to run into snags like this?

Also I edited my preferences as you suggested. The report view also says no module named_coin. It must have failed to build. I am not sure how to take care of this, but it doesn't seem to be specifically related to assembly 2 since I get this error when I open other toolbars. Do you know if this is a common error? Google does not seem to think so.

normandc, 2015-02-07:
Have you installed Ubuntu in a virtual machine? Have you checked that your computer (or virtual machine) supports OpenGL? If you get this error from other workbenches, then it has nothing to do with the Assembly2 module and should be addressed in the FreeCAD forum. http://forum.freecadweb.org/viewforum.php?f=4

Ubuntu 14.10 is fine, but it's only maintained for 9 months which means its support will stop in July 2015.

hamish2014, 2015-02-08:
Hi TimothyStiles, 

FreeCAD and the assembly 2 workbench run fine on my Ubuntu 14.10 desktop.
Perhaps the error has to do with your pivy installation.
Are you abble to import pivy in python concole?
i.e. from bash
$ python
then from inside the Python interpreter
from pivy import coin

TimothyStiles, 2015-02-08:
Hi hamish2014,

I think that you may be on to something. When I tried to run,"from pivy import coin" in the python interpreter it returned, "No module named pivy". I ran "sudo apt-get install python-pivy" and then ran, "from pivy import coin", from terminal bash again and it accepted the command. I tried running the same thing in FreeCAD's python console with no success. It must have something to do with how I am building pivy in FreeCAD. I am currently following the in source make instructions where I set a build parameter to 

$ cmake -DFREECAD_USE_EXTERNAL_PIVY=1 -DCMAKE_BUILD_TYPE=Debug .
$ make

http://www.freecadweb.org/wiki/index.php?title=CompileOnUnix#Compile_FreeCAD

Which builds, but does not allow me to use anything involving coin. When I try to use build with non-external pivy I immediately get this error.

[image]

Any further ideas on what I can do?

normandc: I am running ubuntu 14.10 natively on a mid 2010 Macbook Pro which I know can run openGL. The part design bench works as expected it is just that I cannot get arch, assembly, or spreadsheet to run without throwing the, "No module named coin error." I will post a link to a forum post I made about this when the post has been approved

TimothyStiles, 2015-02-09:
Hey guys,

wmayer helped me solve the problem so feel free to close this issue. (It wasn't really related to assembly2 anyways) the link to the forum post that solved my problem is here for anyone that may have the same problem that I did compiling.

http://forum.freecadweb.org/viewtopic.php?f=4&t=9660

Thanks you guys!

hamish2014, 2015-02-09:
Glad you came right TimothyStiles.
Also thanks for your assistance normandc!

closed 2015-02-09 by hamish2014