hamish2014/FreeCAD_assembly2 #7: Pipe flange assembly.

mange76, 2015-02-12:
Hi! I am assembling pipes with flanges that have boltholes in a circle.I'm using the edge of the bolt holes and circular constraint to connect the pipes.When i connect the first hole everything works fine but when i use the circular constraint for a second hole to align the direction of the pipe i get an error message that The assembly2 solver failed to satisfy the constraint "circularEdgeConstraint02".If i press ignore the pipes seems to align but further assembly usually fail. I can delete the second circular constraint after ignoring and continuing the assembly but pipes are usually rotated out of place when applying new constraints.I also noticed that if i use polar pattern to place the boltholes on the flange the circular constraint completely misses its target so i use a hexagon as a mold to place the circles on the flange and then uses the pocket tool to create the boltholes.

Is there a better workflow to get this done?

I also would like like to say thanks for creating this "lifesaving" WB for us creating larger assemblies:-)

hamish2014, 2015-02-12:
Hi, mange76.

I believe i know what is going on, but would like to check.
Please upload your pipe flange assembly files on dropbox or the like and I will have a look at it.

mange76, 2015-02-12:
Hi! I have uploaded a test assembly file to google drive that you can 
access using the following URL:
  https://drive.google.com/file/d/0BxvNGan2M11wS0FJX1VfWlB0ZW8/view?usp=sharing
I have used the circular constraint to align one of the flanges bolthole 
and then another one to align the rest of the boltholes.The second 
constraint is created after pressing ignore on the error message.I have 
to delete the second constraint to be able to carry on the assembly.
Den 2015-02-12 15:42, hamish2014 skrev:

hamish2014, 2015-02-13:
Hi mange76,

I added support to the solver for cases such as your `align one of the flanges bolthole and then another one to align the rest of the boltholes', so pull to the latest version.
However the numerical solver still experiences quite alot of difficulty, so try different combinations until the solver succeeds. 
(That is why the issue is being left open...)
I have added your pipe assembly as one of testcase for the assembly2 workbench, hope you dont mind.
 

[image]

mange76, 2015-02-14:
Impressive! It now just works most of the time.If it does not work with the circular constraint, i can use a different approach like using the axial constraint for the most cases. I have done some further assembling and added a extra heat exchanger to the assembly.When i connect the first pipe to the exchanger everything works fine but i have not been able to constraint the second pipe.Might it be some rounding errors that makes things a little bit out of alignment? I do not have the knowledge on how to verify if that's the case. You can download the second assembly file from https://drive.google.com/file/d/0BxvNGan2M11wUUxaYkZybWlIMjg/view?usp=sharing if you would like to check it out.

Would it help if i upload the part files for this assembly?

hamish2014, 2015-03-02:
Hi, firstly my apologies for the late reply.

I had a look at your assembly, and could replicate the error.
The numerical solver fails, being unable to solve the constraints to the required accuracy level:
tol [3.379405e-03 > 1.000000e-05]
Here is what is happening inside the solver:

>>> import solverLib
>>> solverLib.analytics['lastSearch'].plot()

[image]

In this case, I believe the solver is failing because because of numerical errors introduced in previous constraints.
In particular, the top flanges and bottom flanges as slightly misaligned, making the constraint for connect the extra heat exchanger impossible
As a work-around I would create an angular constraint between an edge on the extra exchanger and the `seem' edge on the bottom flange.

In the meantime, I am going to think a more general solution.
Perhaps if i reduce the error tolerance's on the constraints, built up errors like yours can be avoided...