hamish2014/FreeCAD_assembly2 #8: Cannot find icon: partsList.svg

wood-galaxy, 2015-02-17:
Hi hamish,
When I load Assembly 2 WB i get this in report view :
Cannot find icon: /home/jo/.FreeCAD/Mod/Dimensioning/partsList.svg
In fact there is the file in FreeCAD_assembly2 folder but not in Dimensioning folder.

hamish2014, 2015-02-18:
stange; 
Please pull the latest version, and see if the patch contained, fixes things

wood-galaxy, 2015-02-18:
Yep it's really strange,
I don't see why the path change to Dimensioning instead of Assembly2.
>>> import os
>>> file = '/home/jo/.FreeCAD/Mod/FreeCAD_assembly2/assembly2lib.py'
>>> os.path.dirname(file)
'/home/jo/.FreeCAD/Mod/FreeCAD_assembly2'

Anyway, I delete all pyc file and now it's working WITH and WITHOUT the path. Very sorry for the noise.
Excellent work Hamish and thank you.

hamish2014, 2015-02-18:
Pleasure, glad it is working again.

closed 2015-02-18 by hamish2014