hamish2014/FreeCAD_assembly2 #9: assambly

ybshaikh, 2015-02-18:
[image]
plz suggest me

ErkanMDR, 2015-02-19:
oh. You are using Windows

hamish2014, 2015-02-19:
Hi ybshaikh,
Is your issue specific to the assembly 2 workbench?
i.e. can you switch to the other workbenches (part, part design, ...) without issue?

sgrogan, 2015-02-23:
I believe this is a problem with the numpy included in the Win dev snapshots.
See this thread on the FreeCAD forum
http://forum.freecadweb.org/viewtopic.php?f=10&t=8395&start=50#p77189
It is specific to XP. Peterl94 posts a link to a numpy version that solves the problem a view posts down from the link (in the same thread)

hamish2014, 2015-03-02:
closing due to lack of activity

closed 2015-03-02 by hamish2014