hamish2014/FreeCAD_assembly2 #11: Problem in Assembly 2 when I change the direction of a planeConstraint

jbet, 2015-02-27:
I have done an assembly with to part.
I don't have any problem with the first assembly ( one planeConstraint, and two axialConstraint).

In the second assembly, I use the same part, but I have change a direction of the planeConstraint.
Then I do the same axialConstraint than the first assembly. And the problem begin, when I put the second axialConstraint.

Conclusion...
Assembly 2 is OK, but if sometime, you change the direction of a planeConstraint, the "problem begin".

I will send you a pdf file, but it is refused by this server.

Best regard.

Jb ETIEVE
France.

hamish2014, 2015-03-02:
Hi Jb,

There are 2 possibilities here:
  1.  a bug
  2.  the direction constraints are contradictory 

Please upload your FreeCAD assembly file onto dropbox or google drive, so that i can determine which.

Warm Regards,

jbet, 2015-03-02:
Thank you for your answer.

OS: Ubuntu 14.04.2 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.15.4537 (Git)
Branch: master
Hash: 79173e37cd0886e197129bf06bfbee01890480b3
Python version: 2.7.6
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.7.1

I have done this assembly with the last version of assembly 2.

I hope you will find this bug.

The file
assemblage, only for information, is what I have doing for my student ( 11 year's old)
It is the second version, last year I have done 10 like that.

The problem :

The file :
croisillon and demi poutre are the original file

The file assembly Ok
 is an assembly where I don't replace the direction of the plane constraint aligne by opposed.

The file assembly non Ok
 is an assembly where I replace the direction of the plane constraint aligne by opposed.
 then I put then constraint, and the error arrive (bug assemblage2.png)

Best regard

----- Mail original -----

closed 2015-08-05 by hamish2014