hamish2014/FreeCAD_assembly2 #12: Problem with plane constraint

Bouni, 2015-03-02:
Hi,

i try to apply a plain constraint to the two parts here:

[image]

I already applied two axial constraints to two of the holes in both elements. If i now apply the plane constraint to the yellow surface and the rear side of the plate, i get the following error message:
```
UNABLE TO SOLVE CONSTRAINTS! info:<AxisDistanceUnion System bearing_mount_import01:Face12-lower_wall_import01:Face9 heirachy 4> abs( self.constraintEq_value(self.X) ) > tol [9.600000e+01 > 1.000000e-05]. Constraint Tree:
<AxisDistanceUnion System bearing_mount_import01:Face12-lower_wall_import01:Face9 heirachy 4> 1 degrees of freedom:
  <AxisRotation DegreeOfFreedom bearing_mount_import01 axis:[  0.00000000e+00   1.00000000e+00  -2.10942375e-15] value:0.000001>
    <AxisAlignmentUnion System bearing_mount_import01:Face12-lower_wall_import01:Face9 heirachy 3> 2 degrees of freedom:
      <AxisRotation DegreeOfFreedom bearing_mount_import01 axis:[  0.00000000e+00   1.00000000e+00  -2.10942375e-15] value:0.000001>
      <LinearMotion DegreeOfFreedom bearing_mount_import01 direction:[  1.00000000e-06  -1.00000000e+00   1.27571535e-15] value:-3.628394>
        <AxisDistanceUnion System bearing_mount_import01:Face11-lower_wall_import01:Face6 heirachy 2> 2 degrees of freedom:
          <AxisRotation DegreeOfFreedom bearing_mount_import01 axis:[  0.00000000e+00   1.00000000e+00  -2.10942375e-15] value:0.000001>
          <LinearMotion DegreeOfFreedom bearing_mount_import01 direction:[  1.00000000e-06  -1.00000000e+00   1.27571535e-15] value:-3.628394>
            <AxisAlignmentUnion System bearing_mount_import01:Face11-lower_wall_import01:Face6 heirachy 1> 4 degrees of freedom:
              <Placement DegreeOfFreedom bearing_mount_import01-x value:-0.000012>
              <Placement DegreeOfFreedom bearing_mount_import01-y value:12.000000>
              <Placement DegreeOfFreedom bearing_mount_import01-z value:65.621017>
              <AxisRotation DegreeOfFreedom bearing_mount_import01 axis:[  0.00000000e+00   1.00000000e+00  -2.10942375e-15] value:0.000001>
                <FreeObjectSystem bearing_mount_import01> 6 degrees of freedom:
                  <Placement DegreeOfFreedom bearing_mount_import01-x value:-0.000012>
                  <Placement DegreeOfFreedom bearing_mount_import01-y value:12.000000>
                  <Placement DegreeOfFreedom bearing_mount_import01-z value:65.621017>
                  <Placement DegreeOfFreedom bearing_mount_import01-azimuth value:0.000000>
                  <Placement DegreeOfFreedom bearing_mount_import01-elavation value:0.000000>
                  <Placement DegreeOfFreedom bearing_mount_import01-rotation value:4.712389>
                <FixedObjectSystem lower_wall_import01> 0 degrees of freedomremoved constraint planeConstraint01
```

Any idea whats the problem here?

And by the way, thank you so much for Assembly2!

hamish2014, 2015-03-02:
Howzit Bouni, 

This looks like a bug.
Please upload your FreeCAD assembly file onto google drive or dropbox  so that I can determine which.

Warm regards,

Bouni, 2015-03-02:
Here are the files:

http://www.filedropper.com/freecad-files-1

hamish2014, 2015-08-05:
Closed due to inactivity

closed 2015-08-05 by hamish2014