hamish2014/FreeCAD_assembly2 #13: I can't load assembly2 workbench

netbomo, 2015-03-11:
Hi,
thaks for your work.

I've just recompiled freecad :
OS: Debian GNU/Linux 7.8 (wheezy)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.15.4643 (Git)
Branch: master
Hash: 9ba57683d59bc57d08b75d56f1c1f05971b922cf
Python version: 2.7.3
Qt version: 4.8.6
Coin version: 3.1.3

and I can't load this workbench, it stop imediatly freecad, in my terminal I have this message :
FreeCAD: SoType.cpp:730: SbBool SoType::isDerivedFrom(SoType) const: Assertion `!this->isBad()' failed.
Aborted

Have you some idea?

gilou

netbomo, 2015-03-11:
Sorry, it's not a problem with your workbench, it's a problem with Coin

gilou

closed 2015-03-11 by netbomo

hamish2014, 2015-03-11:
No worries, hope you manage to sort out the Coin issue.