hamish2014/FreeCAD_assembly2 #14: Assembly from to assembly

jbet, 2015-03-17:
I don't arrive to do an assembly witch contains another assembly.
Is it impossible, or how can we do ?

If it is possible, do a new video for you tube.

Best regard.

leerino, 2015-04-28:
First: I started using the assembly2 workbench and it is a real improvement to Freecad. Thank you for your work.

You can assemble an assembly in an other assembly if you mux the assembly first, then show the muxed assembly and hide the parts assembled in it. While working this way I thougt that it would be great if the the mux feature would be automatically added and showed at the end of each assembly. The user would not even have to see this feature. This way you can automatically update assemblies that contain assemblies. Now I have to remove the mux feature and add it again if I change something.
Another disadvantage now is that while muxing I lose the appearance setting of the assembled parts. Maybe it is possible to mux the parts automatically and assign the appearances from the parts to the surfaces.
And the last difficulty is that the paths are absolute so when you move your assembly you have to upate the paths of the assembly and all subassemblies. Maybe the possibility to make paths relative could solve this.

Best regards.

hamish2014, 2015-04-29:
Hi Leerino,

Thanks for your comments.

I have split you comments into the following parts:
a) importing sub assemblies (shape date only)
b) loosing face colour data when importing parts
c) having the muxed assembly automatically update when the assembly constraints are solved,
d) and that object pushed to the bottom of Gui Object tree
d) absolute paths instead of relative paths

Regarding:
 a) I will see if i can update the import part function to identify assembly2 files, and import the muxed assembly. The muxing will be done automatically.
b&d) I do not know how to do this, and will ask on the FreeCAD forums
c) i have a concern about the speed here. And with the automatic muxing done in a) i do not think this is nessary...
e) i would also prefer relative paths, but do not know a Pythonic and Os independent way of doing this

leerino, 2015-04-29:
Hi Hamish,

Smart idea to put the muxing on the import side. When this is difficult
you could also recreate a mux feature at the end of a file whenever you save an assembly2 document. Then you can just remove the old mux feature and create a new one on save. This way the top assembly does not have to check or remux all subassemblies on recalculation.

leerino, 2015-05-02:
I just tried the new import function and it works very well. A very elegant solution because the muxing is entirely on the background now and in the subassemblies you can directly select the part you want to work on (you don't have to hide or delete the muxing feature first).
Thank you for the quick response and nice solution.

For the paths I was thinking, is there maybe something possible with simple string manipulation...
When a path is relative, you subtract the part of the path that is equal to the assemblies path and store it. When retrieving parts you add the assemblies path to the relative path again. This way all function calls can be with absolute paths that are constructed when needed.

Example:
/mydocs/freecad/project_a/myassy.fcstd

for /mydocs/freecad/project_a/concepts/mypart.fcstd
store: /concepts/mypart.fcstd
add:   /mydocs/freecad/project_a
filename will be:  /mydocs/freecad/project_a/concepts/mypart.fcstd
rename project_a to project_c.
filename will be: /mydocs/freecad/project_c/concepts/mypart.fcstd

for /mydocs/freecad/project_b/mysecondpart.fcstd
store: ../project_b/mysecondpart.fcstd (the dots are just for remembering to remove one directory from the path to add)
add:  /mydocs/freecad
filename will be:  /mydocs/freecad/project_b/mysecondpart.fcstd
rename project_a to project_c.
filename will be: /mydocs/freecad/project_b/mysecondpart.fcstd

f3nix, 2015-08-13:
Hi,
are there any news about the color import?

1. Importing a single part with color works good.
2. Importing assembly (sub) looses color info.
3. Importing an assembly that has been fusioned half works. The whole subassembly gets a color of the first part in the fusion.

Cheers,
Mateusz

hamish2014, 2015-08-16:
Hi Mateusz,

Support has bee added for import face colours when importing part.
I have only had a chance to briefly test this new code, but everything seems ok.
Please try it out and let me know if it works on your side.

f3nix, 2015-08-16:
Hi Hamish2014,
unfortunatelly it is not working quiet as expected.

A small assembly with two elements loads just fine.

A larger assembly makes the FreeCAD process go to 100% CPU and 10% memory and it just hangs there. So after ~10minutes I gave up and killed the process. When I revert back to b96c9e869a0d9ad810e59ff50cb7c94c7f664d18 the import works just great but without color information.

Here You can get the file I'm working with:
https://freeshell.de/~f3nix/freecad/assembly-test5.fcstd

Cheers,
Mateusz

f3nix, 2015-08-16:
Oh,
And when I click mux assembly in the latest version the muxedAssembly part does not retain color information.

Cheers,
Mateusz

hamish2014, 2015-08-16:
Mateusz, this is the best i can do at this moment:
face mapping quicker, however due to what i assume to be numerical noise color mapping fails for certain cases.

f3nix, 2015-08-16:
No worries. I can probably live with that! ;)

Is it possible to add some debug output to see which parts it is choking at?

Cheers,
Mateusz

f3nix, 2015-08-16:
Ok. So here are some more tests.

The slower version 8cc96c820f255ba0cc945ac03f65485347768d8f imported the file without errors. But it takes ~20min on my computer.

The faster version takes ~2min.

But I think I'll stay with the slower version because it imports all faces with proper colors. It will take me longer to update the color of the faces manually.

Here are the screenshots:
"Fast import"

[image]

"Slow import"

[image]

Cheers,
Mateusz

PS. Maybe it is possible to add some kind of progress indicator so one does not think that the program hung?

f3nix, 2015-08-20:
Hi Hamish2014,
thanks for 3ad98c14779919090dee271f8108d9026a82c2f0. It works really good.

Thank You very much! :)

Cheers,
Mateusz

leerino, 2015-08-26:
Hi Hamish,

just tried the latest version and it is working well. Small assemblies import almost instantly and a more complex one in less then a minute. Thank you for that.
I noticed that the transparancy setting is lost. Is it possible to do the same for transparancy?

hamish2014, 2015-08-27:
Hi leerino,

Assuming you asking about maintaining transparent settings for the individual faces of the muxed assembly:
I think it should be possible, but i can't manage it for some reason.
Its seems that each objects face colors in the ViewObject.DiffuseColors attribute has an Alpha component.
However setting that alpha component has no effect...

leerino, 2015-08-28:
Hi Hamish,

Can it be that the alfa value of ViewObject.DiffuseColors is disregarded
and an other property is used? I read in the forum that there is a
ViewObject.Transparency property.

Best regards,

Leerino

leerino, 2015-09-14:
Hi Hamish,

Because relative paths would really be helpfull I decided to dive into Python and came up with a path class that can be used to create relative paths from two absolute paths and absolute paths from a relative path and an absolute path. It can be used as follows:

# Create paths:
my_path = path(my_path_string)
my_base_path = path(my_base_path_string)

# Get the relative path string:
my_relative_path_string = my_path.get_relative_path(my_base_path)
my_relative_path = path(my_relative_path_string)

# Get the absolute path string:
my_absolute_path_string = my_relative_path.get_absolute_path(my_base_path)

# Change the base path:
my_new_base_path = path(my_new_base_path_string)

# Get the new absolute path string
my_absolute_path_string = my_relative_path.get_absolute_path(my_new_base_path)

The base path would be the active assemblies path including the filename. When a relative subpart is placed a relative path can be constructed from the active assemblies path and the subparts path. When you later need to retrieve the subpart you can construct an absolute path with this relative path and the active assemblies path.

The member functions return strings because that is what you need to store the path or retrieve the file. At the moment the class only handles Linux and Windows paths and uses the base path to return a path for the right platform. Also it is not a path validation tool, there are better solutions for that.

I think that with this class it should be possible to add a 'relative path' subpart that stores the relative path/filename and returns the right absolute path/filename when needed. An 'absolute path' subpart can then be used for library parts and the 'relative path' subpart for normal project parts.

If you are interested I can sent you the files.

Best regards,

Onno

hamish2014, 2015-09-19:
Wow.
It is not every day that a user write a library for the assembly 2 workbench.
Thanks Onno!

Unfortunately however I am not going to implement your library:
As a couple of weeks back i added functionality to assembly2 whereby imported part sourcefiles are `repaired' using 'relative paths' if the file cannot be found.

Thanks again however.

closed 2015-09-19 by hamish2014