hamish2014/FreeCAD_assembly2 #16: Issue when importing parts from different file path

zisi, 2015-04-15:
I encountered an issue when I tried to open assembly file that has been created in a different path.

Steps to reproduce
* Create an assembly file in path ``/path/to/file``
* Move to a different path the assemble file with the parts that it is being consisted of
* Open it with FreeCad
* Error encountered
 * ``update of <filename> aborted due source file being missing...``

hamish2014, 2015-04-15:
Hi,

Currently import paths are absolute.
Making the import path relative to the assembly file, would fix your issue.

papamat, 2015-04-16:
Hello, thanks for the fast response. 
   There is a workaround to use relative paths? When you import a new part, source file is an absolute path(like /home/user/folder/partName). When I tried to change that to relative (like ./partName)(the imported part is in the same folder with the first) and hit the update parts button I got the same error as above.
 If you think there is an easy way to fix it, I fill confident to apply for a patch, fill free to send me a snippet.

hamish2014, 2015-04-16:
Hi papamat and zisi,

Please pull the latest patch, and try Update Parts command again.

papamat, 2015-04-16:
OK. Works for me, feel free to close the issue.

hamish2014, 2015-04-16:
Pleasure, closing issue.

closed 2015-04-16 by hamish2014