hamish2014/FreeCAD_assembly2 #17: Constraint within an area

lanciaux, 2015-04-18:
Good afternoon,

I really like your Mod - it has helped me a lot (before I was using Draft move tool a lot ... too much ... really too much).

I have had a case where I wanted a pinhole to stay between a closed area : have you some idea to implement this ?

lanciaux, 2015-04-18:
[image]

hamish2014, 2015-04-20:
Hi lanciaux,

Unfortunately, assembly2 does not support limit constraints.

lanciaux, 2015-04-20:
I do not know you have programmed this mod - I was thinking about using projections on same plane with requiring then to not overlap or having biggest projection UNION smallest projection being still equal biggest projection (in area ?).

Do not worry - as it is, it is already great !

closed 2015-04-20 by lanciaux