hamish2014/FreeCAD_assembly2 #19: Add (import) files with cyrylic names is filed.

scuratovo, 2015-05-01:
When I add to assembly file with name then has cyrylic symbols, Aseembly2 send me error:
---------------------------------------------------------------------------------------------------
importing part from /home/kirill/Рабочий стол/ограничитель.fcstd
Running the Python command 'importPart' failed:
Traceback (most recent call last):
  File "/home/kirill/.FreeCAD/Mod/FreeCAD_assembly2-master/importPart.py", line 92, in Activated
    importedObject = importPart( filename )
  File "/home/kirill/.FreeCAD/Mod/FreeCAD_assembly2-master/importPart.py", line 45, in importPart
    partName = findUnusedObjectName( doc.Label + '_import' )
  File "/home/kirill/.FreeCAD/Mod/FreeCAD_assembly2-master/assembly2lib.py", line 62, in findUnusedObjectName
    while hasattr(FreeCAD.ActiveDocument, objName):

'ascii' codec can't encode characters in position 0-11: ordinal not in range(128)
---------------------------------------------------------------------------------------------------

After rename file ограничитель.fcstd to limiter.fcstd Assembly2 is work.
Can You fix it, please?

hamish2014, 2015-05-04:
Hi scuratovo,

Please pull the latest version, and let me know if it works.

scuratovo, 2015-05-06:
Hello! Thank you for fix and all your work!
All names are supported now!

hamish2014, 2015-05-07:
Pleasure Scuratovo, keep well.

closed 2015-05-07 by hamish2014