hamish2014/FreeCAD_assembly2 #20: Assembly2 32 bit or 64 bit

rasomei, 2015-05-09:
Hi ,

i made some assemblys with the assembly2 mod. Under Freecad 0.15 32 bit it is runing perfectly and under freecad 0.15 64 bit it fails. I can´t reconnect some constraints and making the constraints new doesn`t work. What am I doing wrong?

regards

Ralf

hamish2014, 2015-05-11:
Hi Ralf,

The assembly2 workbench uses the FreeCAD position variables to setup the assembly problem.
To be honest I dont know if these FreeCAD position variables are 32bit or 64bit (single or double precision precision).

The source of your troubles could originate from another source other than the 32-bit -> 64 bit change.
To further understand what is going on, i recommend running the assembly2 workbench tests.
On my Linux system i would do that as follows

$ cd  ~/.FreeCAD/Mod/FreeCAD_assembly2
$ python assembly2solver.py

which will run for a couple of seconds, and if every test was passed will print

*All 15 tests passed, time taken 8.24s*

rasomei, 2015-05-11:
Hi Hamish,
attached the results of the workbench tests in my (very old and slow) windowsmachine:

64 bit fails
Windows 8.1 64bit

c:\Program Files\FreeCAD 0.15\bin>python "C:\Program Files\FreeCad 0.15\Mod\Assembly2\assembly2solver.py"
FreeCAD 0.15, Libs: 0.15R4671 (Git)
Testing assembly 2 solver on assemblies under tests/
tests\testAssembly01.fcstd
Importing project files......
updating properties of planeConstraint1
updating properties of planeConstraint2
updating properties of planeConstraint3
 variableManager.X0 [ 0.          0.          0.          0.          0.          0.          1.97918
 -0.212375   -5.54064    -1.74229996 -0.77800873  3.4906585 ]
solveConstraints: searching for fixed object to begin solving constraints from.
solveConstraints base system: <FixedObjectSystem part1>
  parsing planeConstraint1, type:plane
AxisAlignmentUnion - sys2 <FreeObjectSystem part2>
AxisAlignmentUnion analyticalSolution available: part2 has free rotation.
    v [-0.3376328   0.00555788 -0.94126149], v_ref [ 1.  0.  0.], directionConstraintFlag opposed
    analyticalSolution:  axis [ 0.         -0.99998257 -0.00590461], angle -1.22639544425.
AxisAlignmentUnion Logic "part1": no rotational degrees of freedom ignoring.
AxisAlignmentUnion Logic "part2": reducing from 3 to 1 rotational degree of freedom (2 rotation degrees fixed in defining axis of rotation)
UNABLE TO SOLVE CONSTRAINTS! info:Traceback (most recent call last):
  File "C:\Program Files\FreeCad 0.15\Mod\Assembly2\assembly2solver.py", line 85, in solveConstraints
    constraintSystem = AxisAlignmentUnion(constraintSystem, *cArgs,  constraintValue = constraintObj.directionConstraint )
  File "C:\Program Files\FreeCAD 0.15\Mod\Assembly2\constraintSystems.py", line 77, in __init__
    if debugPrint.level >= 3 : dp('  resulting system:\n%s' % self.str(indent=' '*4, addDOFs=debugPrint.level>3))
  File "C:\Program Files\FreeCAD 0.15\Mod\Assembly2\constraintSystems.py", line 211, in str
    txt = txt + ''.join( [ '\n%s%s' %(indent, d.str('  ')) for d in  self.degreesOfFreedom ] )
  File "C:\Program Files\FreeCAD 0.15\Mod\Assembly2\degreesOfFreedom.py", line 146, in str
    return '%s<AxisRotation DegreeOfFreedom %s axis:%s value:%f>' % (indent, self.objName, self.axis, self.getValue())
  File "C:\Program Files\FreeCAD 0.15\Mod\Assembly2\degreesOfFreedom.py", line 117, in getValue
    raise ValueError, "abs( dotProduct(v, self.axis) ) > %e [error %e]" % (tol, abs( dotProduct(v, self.axis) ))
ValueError: abs( dotProduct(v, self.axis) ) > 1.000000e-07 [error 1.722087e-04]
Failed on tests\testAssembly01.fcstd
_______________________________________________________________
and 32 bit is succsessfull

All 15 tests passed, time taken 22.31s

c:\Program Files (x86)\FreeCAD 0.15\bin>

_______________________________________________________________
could be a double precision windows problem ?

regards

Ralf

hamish2014, 2015-05-14:
Hi again Ralf,

I do not think your issue is a double precision error for the following reasons:
* the errors are too large to result from a 32-bit 64-bit discrepancy
* the error should be less on 64 bit precision compared to 32 bit precision (where your case is the opposite)

Very mysterious.

Have tried testing the numpy build inside the 64 bit FreeCAD.
Open FreeCAD, then in the Python console try:
>>> import numpy
>>> numpy.test()
Running unit tests for numpy
NumPy version 1.8.2
NumPy is installed in /usr/lib/python2.7/dist-packages/numpy
Python version 2.7.6 (default, Mar 22 2014, 23:03:41) [GCC 4.8.2]
nose version 1.3.1
<nose.result.TextTestResult run=4955 errors=0 failures=0>
Ran 4955 tests in 23.990s

OK (KNOWNFAIL=5, SKIP=6)

Note that FreeCAD will stop responding until the test is completed...

rasomei, 2015-05-14:
Hi hamish,
atatched my results running numpy.test on FreeCad64bit:

Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:\Program Files\FreeCAD 0.15\bin\lib\site-packages\numpy\testing\nosetester.py", line 392, in test
    self._show_system_info()
  File "C:\Program Files\FreeCAD 0.15\bin\lib\site-packages\numpy\testing\nosetester.py", line 245, in _show_system_info
    nose = import_nose()
  File "C:\Program Files\FreeCAD 0.15\bin\lib\site-packages\numpy\testing\nosetester.py", line 74, in import_nose
    raise ImportError(msg)
ImportError: Need nose >= 0.10.0 for tests - see http://somethingaboutorange.com/mrl/projects/nose
>>> 
Running numpy.test under FreeCad32bit is the same. Nose is missing on my system.

Meanwhile I tested the assembly2 under ubuntu 64bit (14.04) and it works fine.
I think,I can live with this in the moment, so we could close this issue.

Thanks a lot.

Ralf

hamish2014, 2015-05-15:
Ok, closing the issue.
Please let me know if you do end up figuring out what the problem is.

closed 2015-05-15 by hamish2014