hamish2014/FreeCAD_assembly2 #21: Too liberal constraints on round / cylindrical objects

the-snowwhite, 2015-05-18:
Hi
1.
I'm missing a way to restrict the axis constraint to only be linear maybe by adding a linear constraint that excludes the circular rotating bit, so it complies to a linear bearing 
(which does not rotate).
2.
Also it would be great if if were possible to add a property to restrict the circular edge constraint to be sticky  (non rotating).
____________
I'm working on a linear rail design (pic attached) and find that the animate degrees of freedom function wastes lots of time spinning the rods, bearings, bearing pads, screws etc. when the only animation I hope for:
 is to see the carriage slide along the rods, and everything else to be static

I have found no way to restrict the circular movement of round/cylindrical objects.

Also if i put an axial constraint on both sides of the carriage it dosn't move at all. 
However I was able to make an experiment with only the carriage and 2 rods + axial constraints on both rods to carriage: 

If I fixed the position of the carriage both rods would slide back and forth + rotate also with a plane constraint on both rod ends
If I fixed the position of both/either rod(s) the carriage would not move at all.
Only using 1 axial constraint works except the carriage now also rotates along the rod axis......
_____________
Best wishes
Michael
[image]

hamish2014, 2015-05-21:
Hi Michael,

I made 2 enhancements to assembly2 workbench
   1) sticky axial and circular edge constraints
   2) an animate degrees-of-freedom dialogue has been added, allowing for animation of only selected degrees of freedom.
Try them out, and let me know if they work for you.

Love the pic, by the way.

Sidenote: are you using a power thread to drive your linear rail. If so do know a good place to buy one?

the-snowwhite, 2015-05-22:
Thanks very much Hamish
I will most definitely call it a great improvement, I'm able to assemble 
many more parts without
having to use sub-assemblies (which loose their nice internal color info).

I will return later after I have been out playing to the Copenhagen 
carnival.

I'm using parts from here:
http://www.robotdigg.com/

With these:

http://www.robotdigg.com/product/236/Rapid-Threaded-Rod-Stepper-with-450mm-Tr8*20-Lead-Screw

I can go up to 210 mm/s on 12 Volts with a standard pololu DRV8825 1 / 
32 ustep
on a Beagle bone Black / CRAMPS / Machine kit
combination.

Have a Great evening / day

the-snowwhite, 2015-05-23:
Issue solved .. and more given
Well having played with the new enhancements I must say they work very well for my design. 
I have been able to place 71 parts so far (pic attached) and the solver only takes 5 secs to update after moving a part to a different location. 
The new animation dialog is also a great addition.
I would like to suggest the ability to right-->click and select the constraint / parts associated to a degree of freedom if that's possible.
Thanks
[image]

closed 2015-05-23 by the-snowwhite