hamish2014/FreeCAD_assembly2 #22: New contraint

jbet, 2015-05-31:
Can you create a distance constraint between the edge of a cylinder or arc, and the edge of a solid ?
(distance or tangential)

Best regard

matys4877, 2015-10-08:
Also tangential constraint between sphere and surface or 2 spheres are possible. 

Best regards

lanciaux, 2015-10-28:
This would be fine - I have a case where I would like to setup tangential constraint between several cylinders. I could also implement it myself in your code but did not fully understand the implemented process - if you would document a guide to create "new constraints", maybe it will help enough.

Tangential between 2 cylinders mayve could be solved by finding at east one plane limited to circumference including axis in each cylinder and then solving to find angles of thoses plane in respective objects such a way both planes are coplanar.