hamish2014/FreeCAD_assembly2 #23: Can't create plane constraint between two objects

faber80, 2015-06-03:
I modelled two beams and I tried to constrain two faces with no success with a plane constraint. Is it a bug ?
[image]

hamish2014, 2015-06-03:
Is this what you where trying to achieve?
[image]

From the screen shot, it seems that you are using FreeCAD on Windows.
Are you by any chance using the 64 bit version/build?
If so, then this could be a duplicate of an earlier issue...

faber80, 2015-06-04:
Yes I'm under windows. I'll try under ubuntu x64 then I'll report if it works

faber80, 2015-06-04:
Sorry at the moment I can't try. I'have an UbuntuX64 distro on a virtual machine under windows (VMware) and I have a series of problem with Graphic card and opengl.

faber80, 2015-06-04:
[image]

Same error on Ubuntu 15.04 x64 virtual machine

hamish2014, 2015-06-05:
Ok, please run the following tests and post your results.

#### Test numpy

from the terminal on your Ubuntu machine run (the $ suffix)

```
  $ sudo apt-get install python-nose
  $ python
  >>> import numpy
  >>> numpy.test()
```
Does the test give the OK result?

#### Test the assembly 2 workbench
```
  $ cd ~/.FreeCAD/Mod/FreeCAD_assembly2
  $ python assembly2solver.py
```
Do all the test case succeed?
Does the test script output something like 

*All 16 tests passed, time taken 13.49s*

?

faber80, 2015-06-05:
[image]
[image]

All test passed

hamish2014, 2015-06-05:
Good, so that all works.
- Please provide a step by step break down of how you setup the plane constraint causing the crash.
- Can you define other plane constraints successfully, or do all variants fail.
- could you copy and paste the FreeCAD report view output, detailing the error.
Thanks

faber80, 2015-06-05:
1- I press add plane constrint button
2- I pick a plane face on Beam 1
3- I pick a plane face on Beam 2

This is the report error:
UNABLE TO SOLVE CONSTRAINTS! info:<AxisAlignmentUnion System profilato_100x100_import02:Face7-profilato_100x100_import01:Face18 heirachy 1> no degrees-of-freedom to adjust to satify constraints:
<AxisAlignmentUnion System profilato_100x100_import02:Face7-profilato_100x100_import01:Face18 heirachy 1>
    <FixedObjectSystem profilato_100x100_import01> 0 degrees of freedom
    <FixedObjectSystem profilato_100x100_import02> 0 degrees of freedomremoved constraint planeConstraint01

All costraints don't work.
It seems the two beam have all the six degrees of freedom fixed

hamish2014, 2015-06-05:
Perhaps both Beam 1 and Beam 2 have a fixed position?
Check under the import object properties

faber80, 2015-06-05:
[image]

These are all properties I see. Where can I check if the part have a fixed position?

hamish2014, 2015-06-05:
Hi Faber, check under the data (dati) tab, not the view tab.

faber80, 2015-06-05:
[image]

Sorry I'm a newbie with free cad. 
Fram Data Tab seems the objects have a placement

hamish2014, 2015-06-05:
No worries :)
Look under the **import part** category, and you will see the **fixed Position** attribute.
Make sure only *one of the beams* has `fixed Position = true'

faber80, 2015-06-05:
Ok I checked fixed position to false and now it works! Thank you very much for your support and sorry if I waste your time. I'll extensively test assembly workbench. I worked a lot with CATIA, SOLIDWORKS and PRO-E and I think these module is very important to make FreeCad competitive with professional CAD. Thank you for your work.

hamish2014, 2015-06-05:
Pleasure.
I recommend you update to the latest version of the Assembly 2 workbench.
Specifically, my guess is that you got 2 parts with fixed position, by copying a part with a fixed_Position=True.
In the newer versions of Assembly2, the default behaviour is that copied parts have fixed_Position=False irrespective of the parent part.

closed 2015-06-05 by hamish2014

faber80, 2015-06-05:
I download the version I used from ppa:freecad-community/ppa isn't the right way to get the most updated version?

hamish2014, 2015-06-05:
Hi, yes that is the correct way.
According to my understanding some FreeCAD community members have setup this ppa so that it is linked to the lastest version on GitHub.
I wonder if something is wrong with the ppa update system.