hamish2014/FreeCAD_assembly2 #25: Solver failed to satisfy constraint

samuelezanin, 2015-07-11:
Hello, this week I installed freecad 0.15 and last version of assembly2.
I have an assembly project made with 4 freecad project part.
The project is a box with bottom and top + a battery and a pcb wich goes inside the box.
I align correctly the battery inside the bottom box, then I try to align the pcb inside the second half of box. When I do this I have error of "solver failed to satisfy constraint".
I can assemble the two half part in two projects and then do another project wich merge the two assembly project.
I tried to change the order of assembly in several way but without any luck.
Can you give me an email address where send you my project file so you can test it?
Parts are not complex.
Thanks in advance for help.

hamish2014, 2015-07-12:
Hi  samuelezanin.

I will have a look at your assembly files with pleasure.
Please upload them on a site like http://www.filedropper.com/ and send me the download link.

Kind Regards,

samuelezanin, 2015-07-12:
Il 12/07/2015 18:25, hamish2014 ha scritto:
>
> Hi samuelezanin.
>
> I will have a look at your assembly files with pleasure.
> Please upload them on a site like http://www.filedropper.com/ and send 
> me the download link.
>

Thanks, here is the file:

http://www.filedropper.com/contenitoretag

I started aligning tag with upper cover with 2 axial constraint and a 
plane constraint.
When I tried to place the plain constraint i started getting error.
Then I tried to place battery into the bottom cover with 3 plain constraint.
And finally I tried to align the upper and bottom cover with other 3 
plain constraint.

hamish2014, 2015-07-13:
Thanks for uploading your files samuelezanin, I would approach the assembly as shown in
http://youtu.be/QwUzPT1n_rg

The assembly 2 workbench is a little wonky;
Users need to careful of the numerous pitfalls of the workbench.
Hope you did not get to frustrated trying to assemble your device.

samuelezanin, 2015-07-13:
On 13/07/2015 18:51, hamish2014 wrote:
> Thanks for uploading your files samuelezanin, I would approach the
> assembly as shown in
> http://youtu.be/QwUzPT1n_rg

Thanks, tomorrow morning I will retry.
If you want to publish the video on youtube, can I ask you to remove the
names engraved on cover? This may be a trouble for me.

> The assembly 2 workbench is a little wonky;
> Users need to careful of the numerous pitfalls of the workbench.
> Hope you did not get to frustrated trying to assemble your device.

Ok, assembly can be done in several way, but can I ask you where I made
mistakes? Is there a logi I should follow?

hamish2014, 2015-07-14:
Hi samuelezanin, the youtube video was unlisted.
But to be even safer, i have changed it to a private video (so send me your email if you would like to view it).

I do not look exactly were you went wrong.
From the FreeCAD model tree however i could see that you added all the parts and then tried to assemble them (which in assembly 2 may cause troubles).

My advice is do the assembly 1 part at-a-time:
- Add the part
   > constrain the part
   > check the degrees-of-freedom using animate degrees-of-freedom
Then proceed to add the next part and so on.

samuelezanin, 2015-07-14:
Il 14/07/2015 07:49, hamish2014 ha scritto:
>
> Hi samuelezanin, the youtube video was unlisted.
> But to be even safer, i have changed it to a private video (so send me 
> your email if you would like to view it).
>

Ok, I see it yesterday a few times.
Today I tried again, with your suggest and things goes better.

Thanks a lot.

> My advice is do the assembly 1 part at-a-time:
>
>   * Add the part > constrain the part > check the degrees-of-freedom
>     using animate degrees-of-freedom Then proceed to add the next part
>     and so on.
>

Also with only 2 parts (tag and top cover), I had to change the order of 
constraint.
For now it is ok.

You did a good work.

hamish2014, 2015-07-15:
Pleasure  samuelezanin, closing issue

closed 2015-07-15 by hamish2014