hamish2014/FreeCAD_assembly2 #27: Clones

triplus, 2015-07-12:
When importing parts only single object can be visible. That makes sense as we are talking about importing single part.

What would be nice is to be able to have/create clones. This could be Assembly WB feature or support for Draft Clones could be added if possible (at import part time if Draft Clones are made visible or after):

http://www.freecadweb.org/wiki/index.php?title=Draft_Clone

The purpose of Clones would be to be able to control a group of object from single part file but to have multiple copies positioned at different location in assembly at the same time.

hamish2014, 2015-07-14:
Hi Triplus,

I believe the assembly2 workbench already has this functionality:
In FreeCAD
  1.) switch to the assembly2 workbench
  2.) import a part
  3.) select the part
  4.) right click to open the imported parts pop up menu -> assembly 2 -> duplicate *

*If there is not no assembly 2 item on the pop up menu, try selecting the part again.

triplus, 2015-07-14:
You are correct this is already implemented feature. Don't know how i could overlook this.

Cheers.

closed 2015-07-15 by hamish2014