hamish2014/FreeCAD_assembly2 #28: Spherical surface constraint crash

triplus, 2015-07-12:
I imported two parts. One Sphere and one Box. I added Spherical surface constraint by selecting Sphere and one edge vertex of Box. Everything works as expected unless i click on added constraint in Tree View and get crash.

hamish2014, 2015-07-14:
Hi triplus,

Thanks for posting the issue.
I am unable to replicate the crash.
Which version of FreeCAD are you using?
If you are using the devolpment version, please update to latest release and try again.

triplus, 2015-07-14:
You are correct the problem was on my side. After i deleted FreeCAD .cfg files and updated FreeCAD to latest development release and did a fresh git clone to get FreeCAD_assembly2 WB the crash doesn't happen anymore. I don't know what was the exact reason but it is solved now.

Cheers.

hamish2014, 2015-07-15:
Great, glad you came right.

hamish2014, 2015-07-30:
see http://www.freecadweb.org/tracker/view.php?id=2202

wwmayer, 2016-02-29:
There was a commit from 29. August 2015 which should have fixed this crash. See: https://github.com/FreeCAD/FreeCAD/commit/c3656ad64693105d5d2269142c11b86a18c51be8

hamish2014, 2016-02-29:
thanks wwmayer.

closed 2016-02-29 by hamish2014