hamish2014/FreeCAD_assembly2 #29: Check assembly freeze

scuratovo, 2015-07-13:
Hello Dear Hamish! Your work is a Great. You move freecad to everyone use! Recently, I feel more stable your addon. 
  In last update you are added Check of assembly. But I not know how use it. After click on button my freeCad a hanged.

Thank you for your work. Best regards!

Must I attach my model?

hamish2014, 2015-07-14:
Thanks scuratovo, the FreeCAD teams deverse most of the thanks.
The `check assembly' feature is still very buggy, and i am working on improving it.
So watch this space...

scuratovo, 2015-07-14:
Thank you for your work!

hamish2014, 2015-07-15:
Okay, I have release a patch for the check assembly feature
* cancel button now works
* dialog forced to show
* overlap message also now shows % overlap detected
* boundbox precheck added to increase speed (test conducted showed a 10% speed increase ...)

Take note that the check assembly feature makes use of the Shape.common operation from the Part library.
This incredibly useful/ awesome procedure may occasionally hang for 5 minute then give a failure message such as
> BRep_API: command not done

Unfortunately, these errors are out of my control

scuratovo, 2015-07-15:
Its work on simple models, Thank you!

closed 2015-08-05 by hamish2014