hamish2014/FreeCAD_assembly2 #32: Feature not allowed for constraint

DaveH225, 2015-07-28:
I have two parts in my assembly.  A hub that I created myself and a bearing that I downloaded as an IGES file from the manufacturer.  I imported the IGES file and saved it as a FreeCAD file, then created a new assembly by adding the hub and then the bearing.  When I try to add an axial constraint, (or any constraint), I get the message "Not Allowed: hub_assembly.lm67048_01.Face26".  I can select faces or edges from the hub part that I created, just not from the bearing, even though it recognizes the faces and edges of the bearing.  If I do a STEP file then each part of the bearing is individually displayed, and I can't import it into an assembly.  The IGES part is just one part.

Also, (and this might be part of the same problem), once I've selected the constraint type I want the command won't finish because I can't select the second constraint feature.  I can choose another constraint type, but can't get out of the last command selected.

I'm running the latest FreeCAD (0.16) and Assembly2 packages as of the time of this post.  Hope you can help!

DaveH225, 2015-07-28:
So the second issue I described can be overcome by selecting two faces on the same part, (i.e. the hub), which obviously gives an error and then ends the command.

hamish2014, 2015-07-28:
Hi DaveH225,

Please post your 2 part files on http://www.filedropper.com/ , and i will have a look at what i can do.

DaveH225, 2015-07-30:
Thanks!

Here are the links:

http://www.filedropper.com/hub

http://www.filedropper.com/lm67048

hamish2014, 2015-07-30:
<h1> IGES issue 

Ok, I have had a look at your files, thanks for uploading them:
The lm67084 part is made up of BSpline Surfaces (maybe the IGES format does not directly support cylindrical surfaces...) which do not have an axise.
This is causing your difficulty, as the assembly2 workbench axial (and circular edges) constraint can only be applied between surfaces (or lines) which both have the axis attribute.

I myself have experienced the exact same difficulty when working with IGES files.

When i get a moment i will look into writing some code to approximating an axis from a GeomBSplineSurface object.

In the meantime hang tight.

<h2> selection issue

i will quickly see if can open an task dialog so that users can abort the selection process

hamish2014, 2015-08-04:
Hi  DaveH225,

I believe I have added the code necessary to fix the BSpline Surface problem.
For final testing, I want to check it on your assembly again.
Please re-upload your files (the last file have expired), to that i can check.

Thanks

DaveH225, 2015-08-06:
http://www.filedropper.com/lm67048

http://www.filedropper.com/hub

Files uploaded.  Thanks!

hamish2014, 2015-08-06:
Works on my side!
Please update to the latest version of the assembly2 workbench and see if the update works for you too.

closed 2015-08-26 by hamish2014