hamish2014/FreeCAD_assembly2 #33: Assembly 2 does not load

f3nix, 2015-08-08:
Hi,
after checking out the latest git version I get:

cannot import name fit_plane_to_surface1

And the assembly 2 workbench does not load.

Cheers,
Mateusz

P.S. Using latest FreeCAD git version.

hamish2014, 2015-08-08:
Hi Mateusz,

Weird, could you please check if your FreeCAD_assembly2/lib3D.py updated correctly:
- line 479 should contain the fit_plane_to_surface1 function, and 
- line 488 should be the start of the fit_rotation_axis_to_surface1

f3nix, 2015-08-08:
Hi,
grepping lib3D.py:

`f3nix@bigspider:~/.FreeCAD/Mod/FreeCAD_assembly2$ cat lib3D.py | grep -in plane_to_surface1
479:def fit_plane_to_surface1( surface, n_u=3, n_v=3 ):`

`f3nix@bigspider:~/.FreeCAD/Mod/FreeCAD_assembly2$ cat lib3D.py | grep -in fit_rotation_axis_to_surface1
488:def fit_rotation_axis_to_surface1( surface, n_u=3, n_v=3 ):`

And the FreeCAD output is:

>f3nix@bigspider:~/devel/freecad/FreeCAD_sf_master/build2$ LD_LIBRARY_PATH="/usr/local/lin64/gcc/lib" FreeCAD
FreeCAD 0.16, Libs: 0.16R5334 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2015
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

>Fasteners workbench Loaded
Sheet Metal workbench loaded
cannot import name fit_plane_to_surface1

Cheers,
Mateusz

P.S. Thank You for Your great work. I do really appreciate it!

hamish2014, 2015-08-08:
Pleasure Mateusz.

Hmmm, I have used the assembly2 code on 2 systems since the `fit_plane_to_surface1' update without issue.

I would recommend trying the following
- deleting all the *.pyc files, if that fails
- delete FreeCAD_assembly2 and reclone it.

Hopefully this will fix the issue!

f3nix, 2015-08-08:
Neither of these two solutions have worked for me.

I have to revert to 1658270c928ce0fad7f237fcecadbe8af7c747ea. This is the last one that does not exhibit this problem.

I'm not familiar with python.

How can we debug this further?

Cheers,
Mateusz

f3nix, 2015-08-08:
Ok.
Problem solved! :)

There was a lib3D.pyc in the FreeCAD_drawing_dimensioning.

After deleting it the FreeCAD_assembly2 loads without a problem.

Thanks!

Cheers,
Mateusz

closed 2015-08-08 by f3nix