hamish2014/FreeCAD_assembly2 #34: highlight selected entities used for assembly constraints

paul18fr, 2015-08-14:
Dear Hamesh2014

First of all, many thanks for the great work you're making in the module assembly2.

I'm quite new in using FreeCAD modeler, but I don't think the following feature has ever been implemented in the module yet: 
- when the constraint is created and we would like to come back on it again, it's not possible to see what entities were selected, 
- I mean in the data worksheet, the sub elements 1 and 2 only indicate generic names
- if I click on it, there's no visual indication on the CAD model

Thus, for example, will it be possible to:
- highlight such entities (in wireframe mode , flatlines mode or any other mode) in order to easily identify them?
- add the name of the 2 parts in the 2 SubElement value

This is one suggestion among others I can make, if you think it is relevant of course.

Regards

Paul

f3nix, 2015-08-15:
Hi Paul,
:+1: for this enhancement! :)

You can get away with this simple macro for now.

```python
ob = App.ActiveDocument.getObject("circularEdgeConstraint41")
#ob = App.ActiveDocument.ActiveObject
o1 = ob.getPropertyByName("Object1")
obj1 = App.ActiveDocument.getObject(o1)
o2 = ob.getPropertyByName("Object2")
obj2 = App.ActiveDocument.getObject(o2)
se1 = ob.getPropertyByName("SubElement1")
se2 = ob.getPropertyByName("SubElement2")

Gui.Selection.clearSelection()
Gui.Selection.addSelection(obj1, se1)
Gui.Selection.addSelection(obj2, se2)

Console.PrintMessage("Constraint name: " + str(ob.Name) + "\n")
Console.PrintMessage("Object 1 name: " + str(o1) + "\n")
Console.PrintMessage("Object 2 name: " + str(o2) + "\n")
Console.PrintMessage("Sub Element 1 name: " + str(se1) + "\n")
Console.PrintMessage("Sub Element 2 name: " + str(se2) + "\n")
```

You just have to change the name of the constraint in the first line and then copy and paste it into you python console.

HTH!

Cheers,
Mateusz

paul18fr, 2015-08-15:
Dear Mateusz

Thanks for the answer and for the code; I tested it and I can see at least an highlighted face (the second one is smaller than the previous one); it allows to introduce the fact that in that specific case, wireframe mode is probably the easiest one.

Can I make an additional suggestion:
- you can use 2 different colors - one for each entity
- the entity is highligted when you click on one of the 2 SubElements (data -> properties)

Nevertheless it seems there's an error since the "Console.PrintMessage("Constraint name: " + str(ob.Name) + "\n")" instruction

Nevertheless I a newbie in FreeCAD and I've no skill in Python language, so I can make a relevant feedback.

Regards

Paul

f3nix, 2015-08-16:
Hi Paul,
I've fixed the code in my previous comment. The console output should now work.

I'm just starting my journey with Python and FreeCAD scripting so there is nothing more I can do right now... ;)

Hamish2014 would have to chime in.

I think that changing the two faces (edges, points) colors and making the objects transparent while the constraints is highlighted would be a good idea.

Cheers,
Mateusz

hamish2014, 2015-08-16:
Hi Paul and Mateusz,

Adding a command to highlight selected entities used for assembly constraints is good idea.
I will see if i can set up a command under a constraints pop-up menu as to allow them to do this.

hamish2014, 2015-08-16:
please update to the latest version and try out the command

f3nix, 2015-08-16:
Hi,
after playing around with the selection script (selection observer and not Your changes) I think it does not do the job very well.

Selecting != highlighting

After selecting You loose the ability to edit the constraint. One can change the label only.

I think it should be done moreless like the Onshape people do it. Screenshots attached.

[image]
[image]
[image]

Cheers,
Mateusz

paul18fr, 2015-08-16:
Hamish2014,

After updated the module, I tested it - It seems doing the job in Flat line mode, but we need to proceed in 2 times to see the 2 faces (one of the 2 parts must be in wireframe mode otherwise we cannot see the internal face).

As suggested previously, it'll be better to automatically switch to wireframe mode in order to immediately see the 2 entities, won't be ?

but improvement in a good direction ...

Paul

Paul

[image]
[image]

paul18fr, 2015-08-17:
Hi Hamish2014,

I've been noticing some improvements 'the module was upgraded):
- the 2 parts name is highligthed (or in the current screenshots the part and the sub-assembly) in the combo view (upper left area) -> see screenshot 1
- I do not remeber if it was previously the case, but the names appears as well in the lower left part with "Object1" and "Object2"-> see screenshot 2

Is the following suggestion relevant with a look in the screenshot 3: imagine we a "right-click" on the face name (face4 here) and an "edit" menu, then the 2 parts turn in wireframe mode and the contour line of the face highlights, possible?

In any way, great works you're doing

Thanks 

Paul

Screenshot 1
[image]

Screenshot 2
[image]

Screenshot 3
[image]