hamish2014/FreeCAD_assembly2 #35: [Issue]: Error message when using some macros (Work Features - FCCamera at least)

paul18fr, 2015-08-16:
Hi

Some days ago I noticed that the use of Work Features Macro in conjunction with the  Assembly2 module generates an error (see http://hpics.li/c21395e).

In the forum, mario52 helped me --> it is necessary to disable first the Work Features references in the combo view before importing the part.

Is it known ? otherwise hope it helps

Paul

hamish2014, 2015-08-16:
Hmmm,

I have not used the Work Features Macro so I have not come across this issue.
I will see if i can look into it if I get some time...

paul18fr, 2015-08-16:
Hi hamish2014,

First of all thanks for your answers and your feedback on the suggestions.

The Work features Macro is very interesting especially to anticipate the assembly step, by creating reference planes, holes/axisymetric part axis, points and so on.

A basic example: in complex assemblies and instead of using face + offset feature, one can use a reference plane and add constraint on it; very usefull and powerfull as well to my opinion
(ditto for a point or any entities available in the macro)

Paul