hamish2014/FreeCAD_assembly2 #36: Suggestion in a way to manage constraints in the combo view

paul18fr, 2015-08-16:
Dear Hamish,

I would like to make a suggestion on a way to manage constraints in the combo view.

Indeed I've noticed that the constraints are added chronologically at the bottom of the combo view – graphically it is not linked to the parts ; thus I propose the following management

■▾Part1
 &emsp;- Constraint1 (between part1 & part2)
&emsp;- Constraint2 (between part1 & part3)

■▾Part2
&emsp; - Constraint1
&emsp;- Constraint3 (between part2 & part4)

■▾Part3
&emsp;- Constraint2

■▾Part4
&emsp;- Constraint3

■▾Part x
&emsp;- ......

As you can see, the constraint label appears automatically twice i.e. in each concerned part. By this way, it becomes easier to identify wich parts are involved to the constraint.

In addition, if a constraint is removed or a part is deleted from the assembly, then all the impacted constraints have to be updated at the same time.

(I can propose a variant using a popup in which all the information's are included: type of constraint - orientation - offset - and so on) when it is opened from the part ....

Let me being honnest, I've no idea about the complexity of such request.

If not clear, please tell.

Paul

hamish2014, 2015-08-16:
Hi Paul, 

I also want to arrange the combo tree as you suggest.
Unfortunately though, i have not gotten round to figuring out all the details...

hamish2014, 2015-11-05:
To achieve this functionality I need an additional feature to be implemented in FreeCAD, see
http://forum.freecadweb.org/viewtopic.php?f=10&t=13112

jbet, 2015-12-27:
Hello Hamish,
Thank you for the last toll, Flip the direction of the last constraint add. It will be very useful.

I have a suggestion:
When we click on a constraint, could you:

fcazorla, 2016-06-07:
Hello Hamish and everyone.

I've created a fork on my account with a modification i made some weeks ago on my computer for automatic and more informative labels in constraints. Maybe this suits the need explained by @paul18fr 

Code [here](https://github.com/fcazorla/FreeCAD_assembly2/tree/master-auto-constr-label) and description [here](https://github.com/fcazorla/FreeCAD_assembly2/commit/042e94c05493a8a4ad7e19ef291b13a0e8d1ceda).

Please tell me any issues! Thanks and hope this helps.