hamish2014/FreeCAD_assembly2 #38: cannot import name fit_plane_to_surface1

hvontres, 2015-08-20:
Trying to use the latest version on last night's freecad build I get the folowing error:

Traceback (most recent call last):
  File "/home/henry/.FreeCAD/Mod/FreeCAD_assembly2/importPart.py", line 5, in <module>
    from assembly2lib import *
  File "/home/henry/.FreeCAD/Mod/FreeCAD_assembly2/assembly2lib.py", line 16, in <module>
    from lib3D import fit_plane_to_surface1, fit_rotation_axis_to_surface1
<type 'exceptions.ImportError'>: cannot import name fit_plane_to_surface1
Traceback (most recent call last):
  File "/home/henry/.FreeCAD/Mod/FreeCAD_assembly2/importPart.py", line 5, in <module>
    from assembly2lib import *
  File "/home/henry/.FreeCAD/Mod/FreeCAD_assembly2/assembly2lib.py", line 16, in <module>
    from lib3D import fit_plane_to_surface1, fit_rotation_axis_to_surface1
<type 'exceptions.ImportError'>: cannot import name fit_plane_to_surface1
Traceback (most recent call last):
  File "/home/henry/.FreeCAD/Mod/FreeCAD_assembly2/assembly2lib.py", line 16, in <module>
    from lib3D import fit_plane_to_surface1, fit_rotation_axis_to_surface1
<type 'exceptions.ImportError'>: cannot import name fit_plane_to_surface1
Traceback (most recent call last):
  File "/home/henry/.FreeCAD/Mod/FreeCAD_assembly2/assembly2lib.py", line 16, in <module>
    from lib3D import fit_plane_to_surface1, fit_rotation_axis_to_surface1
<type 'exceptions.ImportError'>: cannot import name fit_plane_to_surface1
Traceback (most recent call last):
  File "/home/henry/.FreeCAD/Mod/FreeCAD_assembly2/assembly2lib.py", line 16, in <module>
    from lib3D import fit_plane_to_surface1, fit_rotation_axis_to_surface1
<type 'exceptions.ImportError'>: cannot import name fit_plane_to_surface1
cannot import name fit_plane_to_surface1

hamish2014, 2015-08-20:
Hi,

I believe this is a repeat of #33 
try deleting all the *.pyc files in the assembly2 directory.

hvontres, 2015-08-20:
Thanks. I had tried deleting the ones in assembly2, but I didn't see the one in drawing_dimensioning, which happend to be first in sys.path.

closed 2015-08-23 by hamish2014