hamish2014/FreeCAD_assembly2 #39: Cannot update parts in assembly

f3nix, 2015-08-20:
Hi,
when trying to update parts imported into assembly I get:

```
Running the Python command 'updateImportedPartsCommand' failed:
Traceback (most recent call last):
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/importPart.py", line 187, in Activated
    importPart( obj.sourceFile, obj.Name )
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/importPart.py", line 61, in importPart
    importUpdateConstraintSubobjects( FreeCAD.ActiveDocument, obj, obj_to_copy )
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/importPart.py", line 485, in importUpdateConstraintSubobjects
    newObjSubElements = classifySubElements( newObject )
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/importPart.py", line 432, in classifySubElements
    catergory = classifySubElement( obj, subElementName )
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/importPart.py", line 407, in classifySubElement
    elif CircularEdgeSelected( selection ):
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/assembly2lib.py", line 268, in CircularEdgeSelected
    if hasattr( edge.Curve, 'Radius' ):

undefined curve type
```

Any idea what is going on? It used to work just fine.

Cheers,
Mateusz

f3nix, 2015-08-20:
Hi,
here are the files that are problematic:
https://freeshell.de/~f3nix/freecad/assembly_bug.7z

The other assembly updates just fine.

Cheers,
Mateusz

paul18fr, 2015-08-21:
Hi Mateusz

Well I tested it (on my Ubuntu with the latest released Assembly2 module) and It seems (but maybe I'm mistaken - some parts may be missing) that it goes right as you can see in the 2 screenshoots (no particular error message even in the python console)

NB: your designs are quite impressive and I'm notocing some good ideas in your file structure ...

Paul

[image]

[image]

f3nix, 2015-08-21:
Hi Paul,
the problem is not in the file loading. It opens without problems for me too.

Here is the thing. I have updated the colors of screws in the appropriate files (fasteners directory). And then when I click on "Update parts imported into assembly" (second icon) I get the aforementioned error message. It shows in the report view (View->Views->Report view).

The result I wanted to achieve was changing the color of the screws from green to something else...

[image]

Cheers,
Mateusz

P.S. These are not my designs. It is an open source 3D printer. Lulzbot TAZ 5. I have just ported some parts from pdf and solidworks.

hamish2014, 2015-08-22:
Hi,

The error occurs because one of the edges in a faster part (M8x35 or something) does not have a Curve attribute.
Which is problematic because when i wrote the edge classification code, I assumed all edge objects would have the Curve attribute.

Anyway, patched it.
Please try it out and see if your issues is resolved.

hamish2014, 2015-08-22:
Ps, if you aren't already try out the the 'Bolt Multiple Circular Edges' command (next to animate assembly icon) which i added on Wed/Thursday.
The command should make things a little quicker...

f3nix, 2015-08-22:
Hi Hamish2014,
thank You. This have solved the problem for me and the assembly files updated without problems. :)

Yes. I have played around with 'Bolt Multiple Circular Edges'. It's a real time saver. Thanks again! :) :+1: 

Cheers,
Mateusz

closed 2015-08-23 by hamish2014