hamish2014/FreeCAD_assembly2 #40: Option to temporary disable the automatic solver update

f3nix, 2015-08-22:
Hi,
I would like to suggest a little enhancement: adding an option to temporary disable the automatic solver update.

Use case:
I'm adding ~100 washers and ~100 bolts to the assembly with 'Bolt Multiple Circular Edges;. That goes rather fast. :) Now I have to fix the "direction Constraint" in some constraints. After every edit the constraint system is being solved. It takes time and I have to wait and look at the screen ;)

What I would like to do is to add this parts, fix the constraints and than hit a button to solve the constraint system.

Thanks in advance for looking into it.

Cheers,
Mateusz

hamish2014, 2015-08-23:
Thanks for the suggestion Mateusz, I will look into it.

hamish2014, 2015-08-24:
Hi Mateusz,

To toggle the auto solving preference:
1. switch to assembly2 workbench
2. edit->preferences->assembly2

This way you can adjust a whole bunch of constraints with out any automatic updating, 
then press the solve assembly button (the Rubik's Cube) to update the system

f3nix, 2015-08-24:
Hi,
this is just awesome! :)

Adding about 50 washers and 50 bolts than fixing the orientation takes less than 15 minutes! :D Instead of ~3h before. I do not want to throw my computer out the window anymore. ;)

Thank You very much. I do really appreciate it.

Cheers,
Mateusz

PS. There is a little glitch with the font size in the preferences tab. Screenshot below:
[image]

closed 2015-09-19 by hamish2014