hamish2014/FreeCAD_assembly2 #41: Possibility to catch pattern features (from the part design module) in order to automatically assembly sub-parts

paul18fr, 2015-08-23:
Hi

Regarding the latest posts (especially the Mateusz ones) where one has to manage dozens of sub-parts such as bolts and so one, I've been thinking it might be interesting to be able to "catch" existing "pattern features" previously created in the part design module.

I mimic here an existing feature I meet in a professional modeler: 
- you create a linear/circular pattern in a part
- when you assemble a bolt for example, the module identifies an existing pattern and asks the user if he wants to automatically add a bolt in all the patterns
- if so, X bolts are assembled in one time

I'm just sharing here a thought

Paul

hamish2014, 2015-08-23:
Interesting idea Paul, I will think on it.
Dont except anything soon:
Currently on the top of my assembly 2 to-do-list is
* fixing the object tree so that the constraints are placed under the objects
* enhancing the solvers speed.

paul18fr, 2015-08-23:
Hi hamish2014,

Thanks for your concern

The main goal has ever been to share an idea (I don't have any idea of the amount of work, so); it'll be for a (next) futur

Many thanks for your great contribution

Paul