hamish2014/FreeCAD_assembly2 #42: Constraint system solved multiple times

f3nix, 2015-08-23:
Hi,
sometimes after adding or modifying a constraint I see something like this in the console:

```
creating planeConstraint31
Constraint system solved in 35.89s; resulting system has 3 degrees-of-freedom
Constraint system solved in 34.78s; resulting system has 3 degrees-of-freedom
Constraint system solved in 16.00s; resulting system has 3 degrees-of-freedom
Constraint system solved in 16.52s; resulting system has 3 degrees-of-freedom
Constraint system solved in 13.71s; resulting system has 3 degrees-of-freedom
Constraint system solved in 13.65s; resulting system has 3 degrees-of-freedom
Constraint system solved in 13.68s; resulting system has 3 degrees-of-freedom
Constraint system solved in 18.08s; resulting system has 3 degrees-of-freedom
Constraint system solved in 18.05s; resulting system has 3 degrees-of-freedom
Constraint system solved in 13.62s; resulting system has 3 degrees-of-freedom
Constraint system solved in 13.66s; resulting system has 3 degrees-of-freedom
Constraint system solved in 13.65s; resulting system has 3 degrees-of-freedom
```

It seems the constraint system is solved multiple times. Should it do that?

Thanks.

Cheers,
Mateusz

hamish2014, 2015-08-24:
No, the (assembly) constraint system solved should only be solved once.
Please provide the steps to follow to create multiple solving occurrence?

f3nix, 2015-08-24:
Hi Hamish2014,
unfortunatelly this issue does not occur all the time. It happens only ocasionally.

But since now I know it is a bug and not the intented behaviour I'll try to be more careful and see what steps are required to reproduce this issue.

Thanks,
Mateusz

hamish2014, 2015-08-29:
The assembly constraint system was being solved multiple times after part updates.
Specifically for each object which was updated the constraint system was solved.
This bug originated from import part proxy objects and should be patched now:
brief testing on my Linux system indicates the issue has been resolved...

f3nix, 2015-08-30:
Hi,
my solve was somehow triggered after changing the plane contraint's parameter. None of the parts of the assembly were updated. I'm 99% sure. I wasn't able to trigger this behavior during last week.

Cheers,
Mateusz