hamish2014/FreeCAD_assembly2 #43: Issue after modifying and updating a sub-part

paul18fr, 2015-08-26:
Dear,

(under Windows at least - It has not been tested under Linux yet), I have the following error after modifying and updating a sub-part (see screenshot)

The study files where uploaded for further investigation (please note due to the website limitation, the original .7z compressed file was renamed with the .png extension)
[image]
[image]

Paul

hamish2014, 2015-08-26:
Hi Paul, 

This looks like a bug.
Let me see if i can quickly patch it ...

paul18fr, 2015-08-27:
Hi

hummm ... I tested it again yesterday night after the macro update on my laptop under Ubuntu and everything seemed to go well.

Today I tried under windows and I new error message appeared; I'm a bit confused (I would like to mean that the behaviour depends on the OS ?)

For informations

Paul

[image]

hamish2014, 2015-08-27:
Hi Pual,

This is rather mysterious, to help us to get the bottom of the issue;
I have changed the assembly2 code to throw a more detailed error message.
So when you have a moment, download the latest version and the post the more detailed error message...

paul18fr, 2015-08-27:
here is the latest message
Paul
[image]

questuk, 2015-08-27:
Hi Hamish,

I am running Windows 7

I am using my 2 files found here http://forum.freecadweb.org/viewtopic.php?f=3&t=12203

Tried your latest update:

regards

Gary

Got the following.

Run Gui init script
Init: Running FreeCADGuiInit.py start script...
Init:   Searching modules...
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Sketcher... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\ReverseEngineering... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Image... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\PartDesign... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Fasteners... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Ship... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Plot... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\FreeCAD_drawing_dimensioning-master... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\OpenSCAD... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Fem... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Spreadsheet... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Start... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Points... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Draft... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Test... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Import... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Complete... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Web... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\FreeCAD_assembly2-master... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Material... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Raytracing... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Mesh... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\MeshPart... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Arch... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Robot... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Part... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Idf(InitGui.py not found)... ignore
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Inspection... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Drawing... done
Init: Loading FreeCAD GUI
Init: Running FreeCADGuiInit.py start script... done
Init: Activating default workbench StartWorkbench
Module: Part
Loading Part module... done
Init: Showing main window
Main window restored
Show main window
Toolbars restored
Init: Entering event loop
Init: Processing command line files
Hide main window
Show main window
Active view is Unnamed : 1[*] (at 0B119810)
Active document is Assembly_of_Jig_015 (at 0DA48B48)
QGLShader::link: "No errors.
" 
QGLShader::link: "No errors.
" 
Loading GUI of Part module... done
updating part Side_support_V1_00_01 from C:/Users/gyb/Desktop/Circular Saw Jig\Side support V1.00.FCStd
Active view is Unnamed : 1[*] (at 0B1272E0)
Active document is Side_support_V1_00 (at 0BEF64E8)
QGLShader::link: "No errors.
" 
QGLShader::link: "No errors.
" 
Loading Sketcher module... done
Loading GUI of Sketcher module... done
Loading PartDesign module... done
Loading GUI of PartDesign module... done
Running the Python command 'updateImportedPartsCommand' failed:
Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.15\Mod\FreeCAD_assembly2-master\importPart.py", line 193, in Activated
    importPart( obj.sourceFile, obj.Name )
  File "C:\Program Files\FreeCAD 0.15\Mod\FreeCAD_assembly2-master\importPart.py", line 30, in importPart
    raise RuntimeError,"doc.FileName != filename (%s != %s)" % str(doc.FileName,filename)

str() takes at most 1 argument (2 given)Active view is Assembly_of_Jig_015 : 1[*] (at 0B119810)
Active document is Assembly_of_Jig_015 (at 0DA48B48)

questuk, 2015-08-27:
Hi Gary again the files are hear

https://www.dropbox.com/s/t1yitv1logq368s/Circular%20Saw%20Jig.zip?dl=0

hamish2014, 2015-08-27:
My apologies:
There was an error in my error message code, x( how embarrassing.
Please download the latest version and try again ...

questuk, 2015-08-27:
Hi,

Latest Assembly2

regards

Gary

Run Gui init script
Init: Running FreeCADGuiInit.py start script...
Init:   Searching modules...
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Sketcher... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\ReverseEngineering... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Image... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\PartDesign... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Fasteners... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Ship... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Plot... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\FreeCAD_drawing_dimensioning-master... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\OpenSCAD... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Fem... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Spreadsheet... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Start... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Points... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Draft... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Test... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Import... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Complete... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Web... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\FreeCAD_assembly2-master... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Material... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Raytracing... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Mesh... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\MeshPart... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Arch... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Robot... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Part... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Idf(InitGui.py not found)... ignore
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Inspection... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Drawing... done
Init: Loading FreeCAD GUI
Init: Running FreeCADGuiInit.py start script... done
Init: Activating default workbench StartWorkbench
Module: Part
Loading Part module... done
Init: Showing main window
Main window restored
Show main window
Toolbars restored
Init: Entering event loop
Init: Processing command line files
Active view is Unnamed : 1[*] (at 0E2AC408)
Active document is Assembly_of_Jig_015 (at 09364C48)
QGLShader::link: "No errors.
" 
QGLShader::link: "No errors.
" 
Loading GUI of Part module... done
updating part Side_support_V1_00_01 from C:/Users/gyb/Desktop/Circular Saw Jig\Side support V1.00.FCStd
Active view is Unnamed : 1[*] (at 0E2ABEF8)
Active document is Side_support_V1_00 (at 10196A08)
QGLShader::link: "No errors.
" 
QGLShader::link: "No errors.
" 
Loading Sketcher module... done
Loading GUI of Sketcher module... done
Loading PartDesign module... done
Loading GUI of PartDesign module... done
Running the Python command 'updateImportedPartsCommand' failed:
Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.15\Mod\FreeCAD_assembly2-master\importPart.py", line 193, in Activated
    importPart( obj.sourceFile, obj.Name )
  File "C:\Program Files\FreeCAD 0.15\Mod\FreeCAD_assembly2-master\importPart.py", line 30, in importPart
    raise RuntimeError,"doc.FileName != filename (%s != %s)" % (doc.FileName,filename)

doc.FileName != filename (C:/Users/gyb/Desktop/Circular Saw Jig/Side support V1.00.FCStd != C:/Users/gyb/Desktop/Circular Saw Jig\Side support V1.00.FCStd)

hamish2014, 2015-08-27:
Thanks Gary, spotted the problem:

Somewhere in *the code* the last '/' gets mangled into a '\':
C:/Users/gyb/Desktop/Circular Saw Jig/Side support V1.00.FCStd
C:/Users/gyb/Desktop/Circular Saw Jig\Side support V1.00.FCStd

Given me sometime while I think of an elegant way of patching this ..

paul18fr, 2015-08-27:
Hi

Module was just updated (Ubuntu) and I've a similar behaviour than the one of yesterday (I still have a small UI error and I need to choose the Assembly module twice / not really distrurbing - this error does not exist under Windows OS).

Please find hereafter an extract of the view report.

Hope it helps

Paul

UI file does not exist
Openning /home/paul/Documents/ETUDE_FREECAD/Devidoir/axe_bobine.FCStd
updating part axe_bobine_01 from /home/paul/Documents/ETUDE_FREECAD/Devidoir/axe_bobine.FCStd
Import: Updating Constraint SubElements Names
  updating axialConstraint01.SubElement1   Face4->Face4
  updating planeConstraint01.SubElement1   Face23->Face23
  updating axialConstraint02.SubElement1   Face35->Face35
  updating planeConstraint02.SubElement1   Face34->Face34
  updating planeConstraint08.SubElement1   Face23->Face23
  updating axialConstraint08.SubElement1   Face24->Face24
  updating planeConstraint09.SubElement1   Face14->Face14
  updating axialConstraint09.SubElement2   Face13->Face13
Constraint system solved in 0.15s; resulting system has 6 degrees-of-freedom
Constraint system solved in 0.15s; resulting system has 6 degrees-of-freedom
Constraint system solved in 0.15s; resulting system has 6 degrees-of-freedom
Constraint system solved in 0.15s; resulting system has 6 degrees-of-freedom
Constraint system solved in 0.15s; resulting system has 6 degrees-of-freedom
Constraint system solved in 0.14s; resulting system has 6 degrees-of-freedom
Constraint system solved in 0.15s; resulting system has 6 degrees-of-freedom
Constraint system solved in 0.15s; resulting system has 6 degrees-of-freedom

hamish2014, 2015-08-29:
Regarding the imported part's source file issue on Windows, I believe the lastest patch should fix the problem:
I have tested the patch on a windows 7 x86 virtual box running FreeCAD 0.15 without issue.
Please test on your side and let me know

questuk, 2015-08-29:
Hi Hamish

Well done !

Works for me on Windows XP

I appreciate your hard work on making this program, I have only been using freecad for a week and your program makes it much easier for a beginner to 'assemble' parts.

Thanks

Gary

questuk, 2015-08-29:
Hi Hamish,

Just noticed something has changed with the update.

Before when i started to create a page, I clicked on 'Open a document or Import files icon' that part became 'fixed position = true' in the property window.

Now when i click on  'import a part from another Freecad document' and place that on the page. This becomes the 'fixed position = true'  is this correct ?

Also these 2 parts will align themselves instantly and not float in free air. I did not make these align they did it themselves ?  (See image)

Not sure if these are bugs, but they didn't do it before.

Regards

Gary

[image]

paul18fr, 2015-08-31:
Hi

I'm a bit confuse but while the update worked fine in my laptop under Ubuntu, it is not the case under my working station under Windows seven (see screenshot).

For feedback

Paul
[image]

paul18fr, 2015-09-04:
Hi,

I don't know if you've ever noticed this, but if I enable the constrain solver, I've some errors dur to the non-update (see picture - still under Windows OS here)

For information

Paul
[image]