hamish2014/FreeCAD_assembly2 #44: Parts not updating in assembly2

questuk, 2015-08-26:
Hi Hamish2014,

I realize you had a similar problem in issue #39 
I seem to have a similar problem (I am a beginner, so it may be me)

If you glance at this thread 
http://forum.freecadweb.org/viewtopic.php?f=3&t=12203

You can see my problem, and i get a python error message, (see the images)

I realize that most parts are missing, but even if delete all other components. Except 'Side support V1.00.FCStd'. From within assembly2, i still get an error message. When doing parts update?

Thanks

Regards

Gary

hamish2014, 2015-08-26:
Hi Gary,

Please post the full python error message given in the FreeCAD Report View Tab,

Thanks

questuk, 2015-08-26:
Run Gui init script
Init: Running FreeCADGuiInit.py start script...
Init:   Searching modules...
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Sketcher... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\ReverseEngineering... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Image... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\PartDesign... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Fasteners... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Ship... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Plot... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\FreeCAD_drawing_dimensioning-master... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\OpenSCAD... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Fem... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Spreadsheet... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Start... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Draft... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Test... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Import... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Complete... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Web... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Material... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Raytracing... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Mesh... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\MeshPart... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Part... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Arch... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\assembly2... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Robot... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Points... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Idf(InitGui.py not found)... ignore
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Inspection... done
Init:      Initializing C:\Program Files\FreeCAD 0.15\Mod\Drawing... done
Init: Loading FreeCAD GUI
Init: Running FreeCADGuiInit.py start script... done
Init: Activating default workbench StartWorkbench
Module: Part
Loading Part module... done
Init: Showing main window
Main window restored
Show main window
Toolbars restored
Init: Entering event loop
Init: Processing command line files
Active view is Unnamed : 1[*] (at 086A6C50)
Active document is Assembly_of_Jig_015 (at 086914D8)
QGLShader::link: "No errors.
" 
QGLShader::link: "No errors.
" 
Loading GUI of Part module... done
Active view is Assembly_of_Jig_015 : 1[*] (at 086A6C50)
Active view is Assembly_of_Jig_015 : 1[*] (at 086A6C50)
Active view is Assembly_of_Jig_015 : 1[*] (at 086A6C50)
updating part Side_support_V1_00_01 from C:/Users/gyb/Desktop/Circular Saw Jig\Side support V1.00.FCStd
Active view is Unnamed : 1[*] (at 06D24FA0)
Active document is Side_support_V1_00 (at 08697E80)
QGLShader::link: "No errors.
" 
QGLShader::link: "No errors.
" 
Loading Sketcher module... done
Loading GUI of Sketcher module... done
Loading PartDesign module... done
Loading GUI of PartDesign module... done
Running the Python command 'updateImportedPartsCommand' failed:
Traceback (most recent call last):
  File "C:\Program Files\FreeCAD 0.15\Mod\assembly2\importPart.py", line 187, in Activated
    importPart( obj.sourceFile, obj.Name )
  File "C:\Program Files\FreeCAD 0.15\Mod\assembly2\importPart.py", line 26, in importPart
    if d.FileName == filename][0]

list index out of range

closed 2015-08-26 by questuk

hamish2014, 2015-08-26:
*if d.FileName == filename][0]* error is a duplicate of #43

questuk, 2015-08-26:
OK thanks Hamish, look forward to the fix.

Regards

Gary