hamish2014/FreeCAD_assembly2 #49: Moving priority

sherif181, 2015-09-01:
There is a small usability problem when you copy an object with ctrl+c and ctrl+v the new part is positioned on the same location as the original part which is alright but when you click on the move tool and then click  on the part the original part is selected  which makes problems if it already has  some constraints.
I lost several hours trying to trick the constraint solver and  I just realized I was  applying constraints to already constrained objects

hamish2014, 2015-09-02:
Sorry to hear about your difficulties.
Rather use the part duplicate command:
1) switch to the assembly 2 workbench,
2) right-click on the part you wish to copy as to launch the pop-up menu
3) assembly2 sub menu -> duplicate part ...

sherif181, 2015-09-02:
It works with copy paste maybe it would help to move the new part a little or  to start  the move command directly after insert .  There is a clone button on drawing I thing  may be it would be good to have a similar button  on  assembly