hamish2014/FreeCAD_assembly2 #52: Asse

jbet, 2015-09-09:
I use Freecad with people 14 and 15 year's old.
I give them the file (s), and they just need to do assembly.

What I need.
I would like to import step file, for assembly.
I use step file, so this people can't modify the part.
Unfortunately, we can't import this sort of file.
Could you modify Assembly 2, for import these file.

Thank you.

hamish2014, 2015-09-10:
Hi, 

Support has been added for directly importing step files, and other formats.
I have only had a brief chance to test the code;
So please pull the latest version and let me know if it works.

jbet, 2015-09-11:
I have tried to import step file, but Assembly 2 want only import .fsctd file.
So I can't import step file.

hamish2014, 2015-09-12:
Hi jbet,

Please check that you are using the latest version of the Assembly 2 workbench.
If you are the option to import a step file will be available: see https://youtu.be/rzlVsTOa9Nw

jbet, 2015-09-13:
OS: Ubuntu 14.04.3 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.15.4671 (Git)
Branch: releases/FreeCAD-0-15
Hash: 244b3aef360841646cbfe80a1b225c8b39c8380c
Python version: 2.7.6
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17
Here is my configuration.

I have update Assembly 2 with:
cd ~/.FreeCAD/Mod/FreeCAD_assembly2
git pull

and I have: Already up-to-date
And when I try, it's does not work as I have seen on Youtube !

Where can I see the number version of Assembly 2 ?
Is the problem come because I use Freecad 15 ?

Best regard

hamish2014, 2015-09-13:
I think the problem may be with the *.pyc files.
Trying removing the them: 
```bash
rm ~/.FreeCAD/Mod/FreeCAD_assembly2/*.pyc
```
do the same if you have the drawing_dimensioning workbench installed
```bash
rm ~/.FreeCAD/Mod/FreeCAD_drawing_dimensioning/*.pyc
```

jbet, 2015-09-14:
I have do it, remove .pyc, and after I get a new update.
But  after this update,  there is the same problem,  it's does not work as I have seen on Youtube !

f3nix, 2015-09-14:
Hi jbet,
it works for me without any problems.

Maybe try to delete the assembly and drawing dimensioning folders and clone them again?

What is the output of these commands?

```
git status
```

and

```
git log
```

[image]

jbet, 2015-09-15:
I delete Assembly 2 and clone it after.
The problem is always the same.

But I have a message when I use Assembly 2 for the first time, 
  UI file does not exist.
At the second time, no problem, Assembly 2 is OK
What that ??????

hamish2014, 2015-09-16:
Please check if you have multiple versions of Assembly2 installed:
I see you are using an Ubuntu system, are you by any chance using the ppa Assembly2 version?
In BASH try
```bash
$ sudo apt-get remove freecad-extras-assembly2
```

jbet, 2015-09-16:
$ sudo apt-get remove freecad-extras-assembly2
OK, very good idea, now all is OK.

I have install ppa Assembly 2, so I know that there is a new version. But Ubuntu does not install it, of course. I use sudo line to install Assembly 2.

Now, I will test if Freecad an Assembly 2 want to work, for step file witch are on the server.

Best regard.

closed 2015-09-16 by jbet