hamish2014/FreeCAD_assembly2 #54: Disturbed by Assembly2 module behaviour

paul18fr, 2015-09-19:
Dear Hamish,

Since yesterday I've been noticing a change in the Assembly2 module behaviour, I feel now that it's affected by external programs/softs involving behaviour changes.

Before I installed a new soft (acroread by the way), (automatic) updates worked fine after creating a new constraint ; now I've troubles when I want to apply a new constraint (plane one) especially if I change the direction (aligned vs opposed) - the update does not worked anymore (while I click onto the icon) and I need to force the calculation ... why???

Can we imagine that external library pollutes the module ?

I reinstalled the libraries numpy and so on using the --reinstall option, ditto for the module bu no change

Does somebody else have ever noticed such trouble?

regards

Paul

hamish2014, 2015-09-23:
Hi Paul,

Although possible that external software has effected the Assembly 2 work-benches behaviour;
it is more likely that auto-solve constraint preference some how got turned off.
To toggle this preference.
  1. switch to the assembly 2 workbench
  2. edit->preferences
  3. assembly 2 -> check or uncheck the auto-solve constraint preference.

Hope this helps

paul18fr, 2015-09-23:
Hi

If I can find it under Windows OS, it does not exist (anymore ?) under my Ubuntu OS - I can not explain whay ?

I need to investigate deeper in order to try ti fix it.

Thanks for your support

Paul

hamish2014, 2015-09-24:
Some how my default preference also changed, see attached gif on how to fix
[image]

paul18fr, 2015-09-24:
Hi Hamish

(ounce again thanks for your support)

Yes I've understood what I need to do, but in my case, the "Assembly2" icon (we can see in your "video" disappeared, and I don't know why ; something should interfered 

Paul