hamish2014/FreeCAD_assembly2 #59: I met a big trouble but it would be nice to know how do you use debugprint ?

lanciaux, 2015-10-22:
In an effort to give you better understanding of trouble (look python related ... python / coin / numpy / scipy /etc.)

How do you catch output of debugprint lines ?

hamish2014, 2015-10-22:
On my Linux system, if I launch FreeCAD from the a BASH Terminal, the debugprint lines are outputted to the terminal:

```bash
 hamish@hamish-pc:~/$ freecad
```

lanciaux, 2015-10-22:
FreeCAD.Console.printmessage works but not debug (should be a DEF to set in BASH ? I do not know it)

Found where it fails but neither it is something I was thinking of : it fails on each second import you do or any update you do of an assembly of at least 2 components

in function  "importpart" of importpart.py

        if filename.lower().endswith('.fcstd'):
            FreeCAD.Console.PrintMessage("Pass Line 39 to open --- %s\n" % (filename))
=> this fails !            doc = FreeCAD.open(filename)
            FreeCAD.Console.PrintMessage("Pass Line 40\n")
        else: #trying shaping import http://forum.freecadweb.org/viewtopic.php?f=22&t=12434&p=99772#p99772

Now I have no idea at all because, I suddenly appears this Sunday after I have updated my Suse 13.2 distro AND FreeCAD I update often. On my other computer with Suse 13.1 at its current level with FreeCAD at same level as in 13.2, it doesn't appear.

This will be difficult to find because this is FreeCAD that is crashing against something in.

We should try to do a simple script that is failling and sending it to FreeCAD team.

Sorry

Ludovic (Belgium)

lanciaux, 2015-10-22:
Maybe this would help : I have loaded 2 times same part

>> doc1 = FreeCAD.open('assembly2/tests/testAssembly16-revolved_surface_objs.fcstd')
>>> App.setActiveDocument("testAssembly16_revolved_surface_objs")
>>> App.ActiveDocument=App.getDocument("testAssembly16_revolved_surface_objs")
>>> Gui.ActiveDocument=Gui.getDocument("testAssembly16_revolved_surface_objs")
>>> doc2 = FreeCAD.open('assembly2/tests/testAssembly16-revolved_surface_objs.fcstd')
Traceback (most recent call last):
  File "<input>", line 1, in <module>
IOError: The project 'assembly2/tests/testAssembly16-revolved_surface_objs.fcstd' is already open!
>>> App.setActiveDocument("Sans_nom")
>>> App.ActiveDocument=App.getDocument("Sans_nom")
>>> Gui.ActiveDocument=Gui.getDocument("Sans_nom")
>>> doc2 = FreeCAD.open('assembly2/tests/testAssembly16-revolved_surface_objs.fcstd')
Traceback (most recent call last):
  File "<input>", line 1, in <module>
IOError: The project 'assembly2/tests/testAssembly16-revolved_surface_objs.fcstd' is already open!

lanciaux, 2015-10-22:
Hard finding again : and if you would commit last updates of FreeCAD it will fail also on Suse 13.1 ... this is well related to fresh modification from FreeCAD maybe this one 
https://github.com/FreeCAD/FreeCAD/commit/969b43bb64b5241ede622a6872b59cabce346fa6

I try to revert to a previous version until it would be fixed

lanciaux, 2015-10-23:
I confirm if you rollback FreeCAD to FreeCAD/FreeCAD@969b43b it works again. Tested on 2 machines today - maybe adding more try catch information can help out how FreeCAD team could help to fix.

Yours

Ludovic

hamish2014, 2015-10-24:
Hi Ludovic,

Thanks for trouble shooting this issue.
The issue also effect my system, which crashes (as you say) on the second part import.
To create a crash

1) open freecad
2) new document
3) assembly 2 import a part
4) assembly 2 import another part

bash output
```bash
$ freecad --log-file /tmp/freecad-log.txt
FreeCAD 0.16, Libs: 0.16R5799 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre 2001-2015
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

importing part from FNAME1
FNAME2 open already False

importing part from FNAME2
FNAME2 open already False
*** Abort *** an exception was raised, but no catch was found.
	... The exception is:SIGSEGV 'segmentation violation' detected. Address 0
```

Let me investigate the issue further and get back to you.

hamish2014, 2015-10-24:
As work around
1) open the PART FILE you wish to import in FreeCAD first, then 
2) import that same PART FILE using the assembly 2 workbench

hamish2014, 2015-10-24:
I agree that this issue likely originates from version 0.16R5799 or FreeCAD.
If the issue continues in later versions, some should file a issue on the FreeCAD bug tracker at
www.freecadweb.org/tracker/

linden1910, 2015-10-25:
glad I am not the only one experiancing this and we have a beter work around than what i have been using.

hamish2014, 2015-10-26:
still crashes in FreeCAD 0.16, Libs: 0.16R5805

hamish2014, 2015-10-27:
Another work around is to use an earlier version of FreeCAD 0.16.
For example the FreeCAD 0.16 version from the 21st of September 2015 (1 month before the issue began occurring) can be compiled on a Linux system with the pre-requisite libraries already installed as follows :

``` bash
$ mkdir FreeCAD_0.16_fallback
$ cd FreeCAD_0.16_fallback/
$ git clone https://github.com/FreeCAD/FreeCAD
$ cd FreeCAD/
$ git checkout 43a4a5938cd1a2438d79726cdaf64ff89e4f774c 
$ cmake .
$ make
```
then to run the fall-back version
``` bash
$ cd bin/
$ ./FreeCAD
```

For additional information on compiling FreeCAD on Linux see http://www.freecadweb.org/wiki/index.php?title=CompileOnUnix

linden1910, 2015-10-28:
thanks very much for all your work developing and supporting this useful 
tool. If you ever need a beta tester or help trying to reprodce bugs let 
me know and I will hel any way I can.

linden

hamish2014, 2015-11-01:
Hi

Does any one know if this issue has been posted on FreeCAD bug tracker ( www.freecadweb.org/tracker/ ) ?

lanciaux, 2015-11-01:
Hello,

not now - I was been waiting for updates of FreeCAD to check if this would be fixed for any other reason. Now, it looks we have to do anyway if we want it to be corrected.

Yours

Ludovic

lanciaux, 2015-11-01:
This is done in FreeCAD bug tracker

hamish2014, 2015-11-02:
Hi Ludovic,

Thanks for all your efforts, particularly in tracking down the Git commit which caused the issue.
Also thanks for posting http://www.freecadweb.org/tracker/view.php?id=2322

As mentioned there, I will investigate the issue at further to respond to
'Unless you can show me there is a bug in freecad, and not in assembly2, I don t see anything we need to do here... '

hamish2014, 2015-11-02:
Good news, after adding a note to http://www.freecadweb.org/tracker/view.php?id=2322 
I had got an idea for a workaround, please pull the latest issue and see if the problem is resolved.

lanciaux, 2015-11-02:
Hello, I'm a bit sad of having worried FreeCAD team yesterday and it's nice your workaround is solving problem as seen of what I have tested so far (from simple ones to intricates ones). I will make test by the end of week on my biggest assembly.

closed 2015-11-02 by lanciaux