hamish2014/FreeCAD_assembly2 #63: Import step file : model automatically saved into same number of sub-parts

paul18fr, 2015-10-25:
Dear

I've imported a STEP file of an assembly containing dozens of sub-parts: how can I (automatically) save the model so that I have not a single file but as many files as the sub-parts I have?

I cannot open a sub-part directly from Assembly2 module and I need to copy/paste the part in a new document: boring and non efficient

NB: it might be interesting to have or not the opportunity either to keep merged the the assembly or to split it into X files

Thanks

Paul

hamish2014, 2015-10-26:
Hi Paul,

Maybe one of the Python Guru's on the FreeCAD forums can help you with a macro to do the task for you.

paul18fr, 2015-10-26:
Hi hamish

First thanks for the answer; I thought it was a good option to split an assembly into the corresponding number of parts (typical topic for an Assembly module), but it does not matter - I'll ask support to the forum

Thanks for your great contribution

Paul

charliestirk, 2016-05-26:
There is a CAX-IF recommended practice that describes how to do assemblies with multiple STEP files and directories.   
https://www.cax-if.org/documents/rec_prac_ext_ref_v21.pdf
FreeCAD cannot open or save STEP file/directories like this, because I believe that the underlying OpenCasCade translator for STEP does not support recommended practices after about 2001.   I recall that Juergen was looking into how to support this with a combination of STEPcode for the assembly structure and OpenCasCade for the geometry with his FreeCAD assembly branch, but I do know the status.