hamish2014/FreeCAD_assembly2 #64: unexpected unsolvablilities during a animation

microelly2, 2015-10-26:
I have done an animation on base of assembly2.
works fine as expected (in most cases)
but I have some unexpected problems
here is the video: https://youtu.be/g04pV5zcaHA

there should not be unsolvable constraints but as you can see sometimes something goes wrong
I have the same problem outside the animation when I  change the arc step by step

I have still no idea how to send you the model because of externel links to the parts.
if neccesary i can make a video how I set up my model 
but at first here is the log:

re init Proxy m01_teil_01
re init Proxy m01_teil_03
re init Proxy m03_kante_01
re init Proxy My_Manager
re init Proxy My_Rotator
re init Proxy My_Tranquillizer
re init Proxy m01_teillang_01
re init Proxy circularEdgeConstraint01
re init Proxy circularEdgeConstraint02
re init Proxy planeConstraint01
My_Rotator
My_Tranquillizer
************************* manager run loop:0/200
Constraint system solved in 0.04s; resulting system has 0 degrees-of-freedom
************************* manager run loop:1/200
Constraint system solved in 0.03s; resulting system has 0 degrees-of-freedom
************************* manager run loop:2/200
Constraint system solved in 0.03s; resulting system has 0 degrees-of-freedom
************************* manager run loop:3/200
Constraint system solved in 0.04s; resulting system has 0 degrees-of-freedom
************************* manager run loop:4/200
Constraint system solved in 0.04s; resulting system has 0 degrees-of-freedom
************************* manager run loop:5/200
Constraint system solved in 0.04s; resulting system has 0 degrees-of-freedom
************************* manager run loop:6/200
Constraint system solved in 0.04s; resulting system has 0 degrees-of-freedom
************************* manager run loop:7/200
Constraint system solved in 0.04s; resulting system has 0 degrees-of-freedom
************************* manager run loop:8/200
Constraint system solved in 0.03s; resulting system has 0 degrees-of-freedom
************************* manager run loop:9/200
Constraint system solved in 0.04s; resulting system has 0 degrees-of-freedom
************************* manager run loop:10/200
Constraint system solved in 0.03s; resulting system has 0 degrees-of-freedom
************************* manager run loop:11/200
[]
Constraint system solved in 0.24s; resulting system has 0 degrees-of-freedom
************************* manager run loop:12/200
[]
Constraint system solved in 0.27s; resulting system has 0 degrees-of-freedom
************************* manager run loop:13/200
[]
Constraint system solved in 0.23s; resulting system has 0 degrees-of-freedom
************************* manager run loop:14/200
[]
Constraint system solved in 0.23s; resulting system has 0 degrees-of-freedom
************************* manager run loop:15/200
[]
Constraint system solved in 0.23s; resulting system has 0 degrees-of-freedom
************************* manager run loop:16/200
[]
Constraint system solved in 0.35s; resulting system has 0 degrees-of-freedom
************************* manager run loop:17/200
[]
Constraint system solved in 0.39s; resulting system has 0 degrees-of-freedom
************************* manager run loop:18/200
[]
Constraint system solved in 0.30s; resulting system has 0 degrees-of-freedom
[... 1139 lines cut ...]
************************* manager run loop:189/200
[]
Constraint system solved in 0.39s; resulting system has 0 degrees-of-freedom
************************* manager run loop:190/200
[]
Constraint system solved in 0.14s; resulting system has 0 degrees-of-freedom
************************* manager run loop:191/200
Constraint system solved in 0.04s; resulting system has 0 degrees-of-freedom
************************* manager run loop:192/200
Constraint system solved in 0.04s; resulting system has 0 degrees-of-freedom
************************* manager run loop:193/200
Constraint system solved in 0.04s; resulting system has 0 degrees-of-freedom
************************* manager run loop:194/200
Constraint system solved in 0.04s; resulting system has 0 degrees-of-freedom
************************* manager run loop:195/200
Constraint system solved in 0.04s; resulting system has 0 degrees-of-freedom
************************* manager run loop:196/200
Constraint system solved in 0.04s; resulting system has 0 degrees-of-freedom
************************* manager run loop:197/200
Constraint system solved in 0.04s; resulting system has 0 degrees-of-freedom
************************* manager run loop:198/200
Constraint system solved in 0.04s; resulting system has 0 degrees-of-freedom
************************* manager run loop:199/200
Constraint system solved in 0.04s; resulting system has 0 degrees-of-freedom
done

hamish2014, 2015-10-28:
Hi microelly2,

The mechanism animated is at the limit of the assembly 2 capabilities.
That is why for certain angles the system solves, well for other angles it fails. 

I have tried myself to animate similar mechanisms, and have found that the solver fails dues to either
* numerical instabalites resulting from the degree-of-freedom variables, or
* numerical errors resulting from floating point operations

Anyway, so although the mechanism looks simple, it is very hard for the assembly2 solver.

see https://github.com/hamish2014/FreeCAD_assembly2/issues/53 for a work around.

microelly2, 2015-10-28:
I hoped that the problem could be fixed when the requests are a little bit softer. 

Can I set  the allowed the tolerance  to 1.0e-03 to avoid this message and get an acceptable animation step

UNABLE TO SOLVE CONSTRAINTS! info: abs( self.constraintEq_value(self.X) ) > tol [2.624588e-07 > 1.000000e-09].

otherwise the idea to solve the system for  one position before and one after the problematic position and interpolate the  parameters in between?
what do you think?

microelly2, 2015-11-17:
I use for this cases interpolation. so I can close the request.

closed 2015-11-17 by microelly2