hamish2014/FreeCAD_assembly2 #65: Assembly2 Import -- SIGSEGV 'segmentation violation' detected.

nathanhack, 2015-10-27:
Hello, I'm using assembly2 (it's pretty good I just wish FreeCAD would fork your work and integrate it, but I digress) and I have a STEP file that it just does not like to import more than once.

I'm using the current FreeCAD. And importing it twice using the built-in (file->import) works fine.  But if i use Assembly to import it twice (into a clean project) it dies with a SIGSEGV.

*** Abort *** an exception was raised, but no catch was found.
	... The exception is:SIGSEGV 'segmentation violation' detected. Address ...

So I started testing if it was only when I imported just that STEP file but it turns out it's not just that STEP file but any that I had.  I was able to get it to not crash if imported it then renamed the label, then import.  I'm not sure where this would be in the code but it seems like an object maybe getting lost into the ether because it shares the same name.

Here's the link to the STEP file I was using https://www.servocity.com/625102.STEP (I would normally just attach it but I do not believe I have the right to do so).

nathanhack, 2015-10-27:
So looks like a dup, sorry.  That said I have a different work around see the above.

hamish2014, 2015-10-27:
Reopened on account of title being more accurate than the title of 59

closed 2015-11-02 by hamish2014