hamish2014/FreeCAD_assembly2 #68: Unable to import a currently open file

Mohamed-Dz, 2015-11-02:
I got this message:
```python
importing part from /home/----/----/----/CAD/SideCap.fcstd
Running the Python command 'importPart' failed:
Traceback (most recent call last):
File "/home/----/.FreeCAD/Mod/FreeCAD_assembly2/importPart.py", line 137, in Activated
importedObject = importPart( filename )
File "/home/----/.FreeCAD/Mod/FreeCAD_assembly2/importPart.py", line 82, in importPart
partName = findUnusedObjectName( doc.Label + '_', document=doc_assembly )

local variable 'doc_assembly' referenced before assignmentimporting part from /home/----/----/----/CAD/ShellAngle.fcstd
importing part from /home/----/----/----/CAD/ShellAngle.fcstd
```
This only happens when I try to import a file that I'm currently editing in FreeCAD, So if I close that file it can then be imported successfully.

**FreeCAD info**
OS: Ubuntu 14.04 LTS
Word size of OS: 32-bit
Word size of FreeCAD: 32-bit
Version: 0.16.5841 (Git)
Build type: None
Branch: master
Hash: e7d16c5f620bc4ddded8b8eae97d6e29c0012490
Python version: 2.7.6
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

**Assembly2 was downloaded today (Nov. 02, 2015).**

hamish2014, 2015-11-02:
Hi Mohamed-Dz,

I pushed through 2 patches today.
The second one, https://github.com/hamish2014/FreeCAD_assembly2/commit/f1e52167d39898832f2fdaef8d60726c4969cda7 , should fix the issue described above

Please pull to the latest version and let me know

Mohamed-Dz, 2015-11-02:
Hello @hamish2014 ;
Yes it works now, Thank you.

closed 2015-11-02 by Mohamed-Dz