hamish2014/FreeCAD_assembly2 #71: Constraints not updated correctly

f3nix, 2015-11-11:
Hi,
I have a problem with the latest assembly. Constraints between two objects are not updated as expected. Here is how to recreate.

1. Create a simple cube.
2. Import it twice into new document.
3. Apply 3 plane constraints so it has 0 dof.

Under cube_01 we have 3 constraints and under cube_02 we have also 3 corresponding constraints. The system has 0 dof.

4. Delete eg. planeConstraint01
5. Hit animate.

The window shows the system has 1 dof. The corresponding constraint from the second part was not deleted (planeConstraint01_). Nor was it taken into consideration while solving the system (dangling constraint?). Clicking "Solve conststraints" does not help either.

Cheers,
Mateusz

PS. I used to group constraints to simulate the movement of some parts of the assembly. Eg. to check the range of motion. With this new grouping scheme it does not seem to be possible. Any suggestions on how to work around it?

hamish2014, 2015-11-11:
Hi Mateusz,

The behaviour described is a bug.
planeConstraint01_ should have also been deleted.
This should be patched now.

For grouping constraints under parts, it was necessary to create 2 objects for each constraint.
These two constraint objects allow a user to modify the constraint from either of part groups.
In the example above this would be
a) planeConstraint01 and 
b) planeConstraint01_

The *_ constraint objects are links/mirrors to the actual constraint objects 
i.e. planeConstraint01_ links to the planeConstraint01, with property modifications of planeConstraint01_ propagating to planeConstraint01.
As such the *_ constraint mirror objects are not actually passed by the solver.

Hmmm, nothing really comes to mind regarding the grouping query.
Perhaps place the parts whose range of motion you want to inspect under a group.

f3nix, 2015-11-11:
Hi,
thanks for that. The delete option works as expected! :)

Would it be possible to add an option in the right click context menu to create a second mirror of a constraint object? It should not be grouped under the object. I could then create a group of 4-5 constraint mirrors that I edit the most and move the assembly elements quickly.

Thanks!

Cheers,
Mateusz

hamish2014, 2015-11-14:
Hi, yes such a modification would be possible.
It would require a lot of work however, since the current code is written for 1 mirror only.
Perhaps yourself or another user can modify the Assembly2 code to your purposes and send through a pull request?

Keep well,

jbet, 2016-01-10:
Hello,
With the latest version of Assembly 2, when I delete a part, all the constraint for this part  are not delete.
I use two mean to delete a part:
     -  select a part in the drawing and press DEL
     -  select  a part in the Combined View, and press DEL.
Could you correct this bug.
I use Freecad and Assembly 2 with pupil 14 year's old, at school.

Best regard.