hamish2014/FreeCAD_assembly2 #73: Request - make assembly solver non  intreactive and silent on demand.

microelly2, 2015-11-17:
I want to integrate assembly2 to the animatio nworkbench

can you add please some modifications

(1) no interaction if the assembly server doens find a solution
#####################################
# hack line 136  assembly2solver.py

if FreeCAD.ParamGet('User parameter:/Plugins/assembyl2').GetBool('nointeractive'):
        response=False
        solved=True
else:
        response = QtGui.QMessageBox.critical(QtGui.qApp.activeWindow(), "Solver Failure!", message, flags)
        response=False
        solved=True

# hack
#############################

and 
(2) it should be  possible to run the server without log output to console on demand

if FreeCAD.ParamGet('User parameter:/Plugins/assembyl2').GetBool('silent'):
        do not write loging information ...

if the two parameters are not set or are false, then assembly2 works as usual
otherwise I can use it in backround

hamish2014, 2015-11-18:
Sure thing microelly2, 
I will look into making the requested changes next weekend.

hamish2014, 2015-11-28:
Alright, regarding suppressing the report view output from the Assembly2 solver do the following in the python console

```python
import assembly2lib
assembly2lib.debugPrint.level = 0 #the default is 2
```

For the Assembly2 solver failing quietly, the following Python code should do the trick
```Python
import assembly2lib
assembly2lib.debugPrint.level = 0
import assembly2solver
constraintSystem = assembly2solver.solveConstraints( FreeCAD.ActiveDocument, showFailureErrorDialog=False, printErrors=False )
if constraintSystem == None:
    print('Solver failed to satisfy specified constraints')
else:
    ...

```

microelly2, 2015-12-07:
Thank you, I have added a Assembly2Controller to Animation WB.
It works now as expected.
https://github.com/microelly2/Animation/commit/8d5b20c9f0364be29d21394e619d3173cfb5b0b6

closed 2015-12-07 by microelly2