hamish2014/FreeCAD_assembly2 #74: A lot of warning or errors (I don't know)

paul18fr, 2015-11-22:
Hi

I've the following warninf or errors (I don't know) when I open my assembly (under FreeCAD 0.16 - daily release); furthermore, The combo view looks strange when I import a sub-assembly.

I can share the complete file if you think it can be usefull - please send an email in a gmail account (nickname at gmail dot com)

Paul
[image]

<pre>
<span style="color: red">
Traceback (most recent call last):
  File "/home/paul/free-cad-code/freecad-build/Mod/Assembly2/viewProviderProxies.py", line 180, in onChanged
    if getattr(constraintObj, prop) != getattr( obj, prop):
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'SubElement1'
Traceback (most recent call last):
  File "/home/paul/free-cad-code/freecad-build/Mod/Assembly2/viewProviderProxies.py", line 180, in onChanged
    if getattr(constraintObj, prop) != getattr( obj, prop):
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'SubElement2'
Traceback (most recent call last):
  File "/home/paul/free-cad-code/freecad-build/Mod/Assembly2/viewProviderProxies.py", line 181, in onChanged
    setattr( constraintObj, prop, getattr( obj, prop) )
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'Type'
Traceback (most recent call last):
  File "/home/paul/free-cad-code/freecad-build/Mod/Assembly2/viewProviderProxies.py", line 180, in onChanged
    if getattr(constraintObj, prop) != getattr( obj, prop):
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'directionConstraint'
Traceback (most recent call last):
  File "/home/paul/free-cad-code/freecad-build/Mod/Assembly2/viewProviderProxies.py", line 180, in onChanged
    if getattr(constraintObj, prop) != getattr( obj, prop):
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'lockRotation'
Traceback (most recent call last):
  File "/home/paul/free-cad-code/freecad-build/Mod/Assembly2/viewProviderProxies.py", line 180, in onChanged
    if getattr(constraintObj, prop) != getattr( obj, prop):
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'offset'
Traceback (most recent call last):
  File "/home/paul/free-cad-code/freecad-build/Mod/Assembly2/viewProviderProxies.py", line 180, in onChanged
    if getattr(constraintObj, prop) != getattr( obj, prop):
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'SubElement1'
Traceback (most recent call last):
  File "/home/paul/free-cad-code/freecad-build/Mod/Assembly2/viewProviderProxies.py", line 180, in onChanged
    if getattr(constraintObj, prop) != getattr( obj, prop):
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'SubElement2'
Traceback (most recent call last):
  File "/home/paul/free-cad-code/freecad-build/Mod/Assembly2/viewProviderProxies.py", line 181, in onChanged
    setattr( constraintObj, prop, getattr( obj, prop) )
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'Type'
Traceback (most recent call last):
  File "/home/paul/free-cad-code/freecad-build/Mod/Assembly2/viewProviderProxies.py", line 180, in onChanged
    if getattr(constraintObj, prop) != getattr( obj, prop):
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'angle'
Traceback (most recent call last):
  File "/home/paul/free-cad-code/freecad-build/Mod/Assembly2/viewProviderProxies.py", line 180, in onChanged
    if getattr(constraintObj, prop) != getattr( obj, prop):
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'SubElement1'
Traceback (most recent call last):
  File "/home/paul/free-cad-code/freecad-build/Mod/Assembly2/viewProviderProxies.py", line 180, in onChanged
    if getattr(constraintObj, prop) != getattr( obj, prop):
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'SubElement2'
Traceback (most recent call last):
  File "/home/paul/free-cad-code/freecad-build/Mod/Assembly2/viewProviderProxies.py", line 181, in onChanged
    setattr( constraintObj, prop, getattr( obj, prop) )
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'Type'
Traceback (most recent call last):
  File "/home/paul/free-cad-code/freecad-build/Mod/Assembly2/viewProviderProxies.py", line 180, in onChanged
    if getattr(constraintObj, prop) != getattr( obj, prop):
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'angle'
Traceback (most recent call last):
  File "/home/paul/free-cad-code/freecad-build/Mod/Assembly2/viewProviderProxies.py", line 180, in onChanged
    if getattr(constraintObj, prop) != getattr( obj, prop):
[... 226 lines cut ...]
  File "/home/paul/free-cad-code/freecad-build/Mod/Assembly2/viewProviderProxies.py", line 180, in onChanged
    if getattr(constraintObj, prop) != getattr( obj, prop):
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'offset'
Traceback (most recent call last):
  File "/home/paul/free-cad-code/freecad-build/Mod/Assembly2/viewProviderProxies.py", line 180, in onChanged
    if getattr(constraintObj, prop) != getattr( obj, prop):
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'SubElement1'
Traceback (most recent call last):
  File "/home/paul/free-cad-code/freecad-build/Mod/Assembly2/viewProviderProxies.py", line 180, in onChanged
    if getattr(constraintObj, prop) != getattr( obj, prop):
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'SubElement2'
Traceback (most recent call last):
  File "/home/paul/free-cad-code/freecad-build/Mod/Assembly2/viewProviderProxies.py", line 181, in onChanged
    setattr( constraintObj, prop, getattr( obj, prop) )
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'Type'
Traceback (most recent call last):
  File "/home/paul/free-cad-code/freecad-build/Mod/Assembly2/viewProviderProxies.py", line 180, in onChanged
    if getattr(constraintObj, prop) != getattr( obj, prop):
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'directionConstraint'
Traceback (most recent call last):
  File "/home/paul/free-cad-code/freecad-build/Mod/Assembly2/viewProviderProxies.py", line 180, in onChanged
    if getattr(constraintObj, prop) != getattr( obj, prop):
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'offset'
</span>
</pre>

hamish2014, 2015-11-28:
Howzit Paul,

Firstly, i dig your spider robot, epic!

Regarding the 'FeaturePython' object has no attribute 'offset' error messages i have released a patch for it which should help.

Regarding the comboview looking strange:
When migrating an assembly file from the pre-grouped assembly2 to the grouped comboview assembly2, try the following

1.) open the assembly file
2.) press the update assemby file (this will create the constraint mirrors), but this mirrors will not be grouped however. Therefore finally
3.) Save and reopen the assembly2 file, after which the grouping should be done...

paul18fr, 2015-11-29:
Hi Hamish2014

First of all, thanks for your support.

I'm not sure to well understand because nothing I've done work (I probably misunderstood your suggestion) - so please find in attachement the complete project( <b>WARNING : note that the .7z extension has been changed to .pdf to be accepted by the website</b>)

The cad files will be shared when the troubles will be fixed, some comments directly in the CAD files added and the dimensions validated (parts are currently 3D printed)

Hope it helps

Regards

Paul
[spider_robot.pdf](https://github.com/hamish2014/FreeCAD_assembly2/files/46753/spider_robot.pdf)

f3nix, 2015-11-29:
Hi,
I can confirm that something strange is going on.

Here are the messages:
`Traceback (most recent call last):
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/viewProviderProxies.py", line 182, in onChanged
    setattr( constraintObj, prop, getattr( obj, prop) )
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'Type'
Traceback (most recent call last):
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/viewProviderProxies.py", line 182, in onChanged
    setattr( constraintObj, prop, getattr( obj, prop) )
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'Type'
Traceback (most recent call last):
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/viewProviderProxies.py", line 182, in onChanged
    setattr( constraintObj, prop, getattr( obj, prop) )
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'Type'
Traceback (most recent call last):
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/viewProviderProxies.py", line 182, in onChanged
    setattr( constraintObj, prop, getattr( obj, prop) )
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'Type'
Traceback (most recent call last):
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/viewProviderProxies.py", line 182, in onChanged
    setattr( constraintObj, prop, getattr( obj, prop) )
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'Type'
Traceback (most recent call last):
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/viewProviderProxies.py", line 182, in onChanged
    setattr( constraintObj, prop, getattr( obj, prop) )
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'Type'
Traceback (most recent call last):
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/viewProviderProxies.py", line 182, in onChanged
    setattr( constraintObj, prop, getattr( obj, prop) )
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'Type'
Traceback (most recent call last):
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/viewProviderProxies.py", line 182, in onChanged
    setattr( constraintObj, prop, getattr( obj, prop) )
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'Type'
Traceback (most recent call last):
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/viewProviderProxies.py", line 182, in onChanged
    setattr( constraintObj, prop, getattr( obj, prop) )
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'Type'
Traceback (most recent call last):
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/viewProviderProxies.py", line 182, in onChanged
    setattr( constraintObj, prop, getattr( obj, prop) )
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'Type'
Traceback (most recent call last):
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/viewProviderProxies.py", line 182, in onChanged
    setattr( constraintObj, prop, getattr( obj, prop) )
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'Type'
Traceback (most recent call last):
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/viewProviderProxies.py", line 182, in onChanged
    setattr( constraintObj, prop, getattr( obj, prop) )
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'Type'
Traceback (most recent call last):
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/viewProviderProxies.py", line 182, in onChanged
    setattr( constraintObj, prop, getattr( obj, prop) )
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'Type'
Traceback (most recent call last):
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/viewProviderProxies.py", line 182, in onChanged
    setattr( constraintObj, prop, getattr( obj, prop) )
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'Type'
Traceback (most recent call last):
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/viewProviderProxies.py", line 182, in onChanged
    setattr( constraintObj, prop, getattr( obj, prop) )
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'Type'
Traceback (most recent call last):
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/viewProviderProxies.py", line 182, in onChanged
    setattr( constraintObj, prop, getattr( obj, prop) )
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'Type'
Traceback (most recent call last):
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/viewProviderProxies.py", line 182, in onChanged
    setattr( constraintObj, prop, getattr( obj, prop) )
<type 'exceptions.AttributeError'>: 'FeaturePython' object has no attribute 'Type'`

Even after updating the sub-assemblies.

Moreover some information is lost because the legs look like this:
[image]

Cheers,
Mateusz

paul18fr, 2015-11-30:
thanks Mateusz,

While I did some changes before, I've not noticed constraint issues so far ; but for me it looks different (I'm using both the latest FreeCAD and Assembly daily modules).

I'll have a look on it as soon as possible

Paul

paul18fr, 2015-11-30:
well I'm sceptic:
- this morning the assembly ans sub-assemblies were updated in my workstation under Win. OS -> I've a constraint issue but not the same order as you had
- the evening I opened the original assembly/sub-assemblies on my laptop under ubuntu I asked for updating but I've no issue except the ones in the first message

So I do know how to say ...

Paul

hamish2014, 2015-12-02:
Hi All,

The report view error message occur during loading.
Unfortunately, I have not figured exactly how FreeCAD loads documents.
In particular in what sequence FreeCAD assigns attributes and calls onChanged methods during loading, is a mystery to me.
Anyway, I have released another patch, which should squelch these kinds of error messages.

No worries Paul, I also occasionally experience weird issues.
Hope things continue to behave.
[image]

paul18fr, 2015-12-02:
Hi Hamish

Your work is really impressive; I cannot help you in developing Assembly 2 module, but I can share my experience in its using and in highlighting some issues (that can be fixed or not)

Hoping my little contribution can help you in improving your module (furthermore I've a lot of projects that include Freecad modeling and the parts assemblies)

Paul

paul18fr, 2015-12-02:
I've fixed all the lost references using "Redifine" tool (hopefully we have such option), nevertheless I cannot exclude another issues in a next future when additional references will be lost and/or faces/edges name will be internally changed

Now 3D printing stage is in progess - I hope the "beast" will walk soon :-)

hamish2014, 2015-12-03:
Thanks Paul.
Good luck with your Spider Beast Robot!

closed 2015-12-03 by hamish2014