hamish2014/FreeCAD_assembly2 #76: Assembly 2 Installation Problem

Clive-D, 2015-12-01:
I'm a new user of FreeCAD. I installed 'Assembly 2' as outlined but it does not show up as a workbench entry in the drop down menu.
(I've installed 'Fasteners' workbench using the same method and it works fine.)
Do you have any suggestions as to what I'm doing wrong with my installation?
C:\Program Files\FreeCAD 0.15\Mod\assembly2 (69 files; 7 folders; 6.81 MB)
I don't fully understand what "Pyside and Numpy are integrated in the FreeCAD 0.15 dev-Snapshots, so these Python packages do not need to be installed individually" means but assume that my setup complies.
Thanks,
Clive

System info:
OS: Windows 7
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.15.4671 (Git)
Branch: releases/FreeCAD-0-15
Hash: 244b3aef360841646cbfe80a1b225c8b39c8380c
Python version: 2.7.8
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17
[image]

f3nix, 2015-12-02:
Hi,
are you getting any error messages in the report view? View->Views->Report View

I think You should move the files one level up. Move all files and folders from FreeCAD_assembly2-165242ca25e2a1daa960005131ff8f859b5be536 to assembly2.

HTH :)

Cheers,
Mateusz

Clive-D, 2015-12-02:
Thanks for the reply. 
Report view contains no error messages. When I restructured the 'assembly2' folder as you suggested, problem solved! 
I think it would be beneficial to amend the installation instruction or the file structure within the zip file to prevent this issue with other newbies. 
Thanks again, Clive

hamish2014, 2015-12-03:
Thanks for the assist Mateusz.

Clive, I will close the issue for the moment.
If other users also experience difficulty i will rewrite the installation instructions.

closed 2015-12-03 by hamish2014