hamish2014/FreeCAD_assembly2 #77: automatically solve when constraints change option doesn't stay

QiangF, 2015-12-02:
Hello everyone,
The problem is the automatically solve when constraints change option changes back to unticked unpredictably. Is there a way to make that option stay ticked?
Thank you!

f3nix, 2015-12-06:
Hi,
I can confirm it.

I have tracked it down to the Activated function in importParty.py (line 186).

Line 190
```         parms.SetBool('autoSolveConstraintAttributesChanged', False)```

If anything after it goes wrong (eg. when loading old assembly which constraint objects have no icon thus not being able to create the mirror object)

```
Running the Python command 'updateImportedPartsCommand' failed:
Traceback (most recent call last):
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/importPart.py", line 254, in Activated
    obj.ViewObject.Proxy.mirror_name = create_constraint_mirror(  obj, obj.ViewObject.Proxy.iconPath )

'int' object has no attribute 'iconPath'
```
It never reaches:

Line 260
```        parms.SetBool('autoSolveConstraintAttributesChanged', org_setting )```

Not sure how to fix it though.

Cheers,
Mateusz

hamish2014, 2015-12-07:
Wow.
Excellent working tracking down the bug Mateusz.

I have written a patch to address the issue:
Instead of altering the FreeCAD preference to prevent multiple solveConstraints being called when updating an assembly,
Imported object's are now purgeTouched to prevent their Proxy.execute methods from being called during an assembly update.

f3nix, 2015-12-10:
Hi hamish2014,
Thank You very much. It does work as expected.

And I have learned something new! :)

Cheers,
Mateusz

closed 2015-12-18 by hamish2014