hamish2014/FreeCAD_assembly2 #81: Can't update old assembly files

f3nix, 2015-12-10:
Hi,
I have a few assemblies created in earlier versions of Assembly2.

I can't update the parts in them.

The error message is:

```
Running the Python command 'updateImportedPartsCommand' failed:
Traceback (most recent call last):
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/importPart.py", line 248, in Activated
    obj.ViewObject.Proxy.mirror_name = create_constraint_mirror(  obj, obj.ViewObject.Proxy.iconPath )

'int' object has no attribute 'iconPath'
```

These assemblies were created before introduction of constraints icons. Is there an easy way to fix it?

Cheers,
Mateusz

hamish2014, 2015-12-16:
Hi Mateusz,

What needs to be done is to loop over all the old constraint objects and assign ConstraintViewProviderProxy's to their ViewObject proxies.
See the code added in patch 7c8fa38 

Doing this work yourself should not be necessary with latest patch.
Simply load the old assembly, and press the solve assembly button.
This will end up calling updateObjectProperties function, which (with the latest patch) will add ConstraintViewProviderProxy to old constraint objects.

Kind Regards,

f3nix, 2015-12-22:
Hi,
Thanks for coding it! I just got some time to test this.

I have spotted one problem. After clicking solve it seems that the update did not work. Mirrors are created but the old constraints look the same (default icon). I have to save, close and reload the document for changes to take effect. This is on latest dev branch.

Any idea what is going on?

Thanks again! :)

Cheers,
Mateusz

hamish2014, 2015-12-23:
Hi,

On my side I also have to save, and reopen the file for the changes to take effect.
Also don't know why :)

Anyway ...

closed 2015-12-23 by hamish2014

f3nix, 2015-12-26:
Hi,
after more testing I think this issue is not solved completly.

E.g.
I'm trying to convert my old assembly. It can be downloaded here: http://forum.freecadweb.org/viewtopic.php?f=24&t=13374

In this assembly parts and constraints were grouped.

This is how it looks in previous assembly (f1e52167d39898832f2fdaef8d60726c4969cda7):
[image]

This is how it looks in latest assembly (after solving and saving the file: 7d71abaa2e24cbb69df050b0029fa72b412b183c)
[image]

And this is how it looks during loading in latest assembly:
[image]

It looks like the objects were not "physically" moved. But the grouping is updated during loading. It introduces a bug.
After loading the assembly and deleting e.g. the "Constrains" group (which seems to be empy) the assembly is no longer solvable.

Hitting solve gives:
```
UNABLE TO SOLVE CONSTRAINTS! info:Traceback (most recent call last):
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/assembly2solver.py", line 106, in solveConstraints
    if constraintObj.lockRotation: constraintSystem =  LockRelativeAxialRotationUnion(constraintSystem,  *cArgs, constraintValue = 0)
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/constraintSystems.py", line 75, in __init__
    self.init2()
  File "/home/f3nix/.FreeCAD/Mod/FreeCAD_assembly2/constraintSystems.py", line 846, in init2
    assert isinstance(sys.sys2, FreeObjectSystem)
AssertionError
```

Cheers,
Mateusz