hamish2014/FreeCAD_assembly2 #82: Last Update of assembly2

IcarePetibles, 2015-12-17:
Hi,
In the latest update (12/16/20015) of assembly2, I can't change in properties:
- direction Constraint,
- offset.
Do you have a solution for this problem?
Best Regards
icare

IcarePetibles, 2015-12-18:
Hi,
Same problem with update of 12/18/2015.
I use:
OS: Ubuntu 14.04.3 LTS
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.16.6066 (Git)
Build type: None
Branch: master
Hash: 24c60199cce8c8def9b67d17cf7569b57fcb3aed
Python version: 2.7.6
Qt version: 4.8.6
Coin version: 4.0.0a
OCC version: 6.8.0.oce-0.17

hamish2014, 2015-12-18:
Hi, 

When you say you can't change the direction of a constraint, or offset, do you mean
a) you can't change these property values, or
b) that changing these properties has no effect

if b) please check your assembly2 preferences are checked as in the attached screen shot
[image]

*NB - change to assembly2 workbench preferences, switch to the workbench first. 
Otherwise the assembly2 prefrences won't be visible under Edit > Preferences

IcarePetibles, 2015-12-18:
Hi,
It was b) and "Automatically solve system when a constraint's property is changed" was not checked.
Thank you for solving of my problem. 
Assembly2 is a great workshop that help me a lot in the design of my CNC milling
Best Regards

closed 2015-12-23 by hamish2014